TriForceX/MiyooCFW #107: Power off shortcut on V90?

MaciekBaron, 2020-09-27:
Hi there,

First of all thank you for all your amazing work!

I was wondering whether there is a shortcut on the V90 that allows you to power off safely. The original firmware wouldn't mind if you switch the console off, while the custom firmware does a disk scan if it detects unsafe power off.

The V90 does not have a power button but rather a power switch.

TriForceX, 2020-09-27:
Have you tried to then Have you tried holding down the reset button? (the button that is between the dpad and ABXY)

MaciekBaron, 2020-09-27:
Thanks for your reply! That triggers "suspend", and when I switch the power off it still causes an unclean shutdown.
But this encouraged me to check other buttons, and it turns out START does the trick, but unfortunately only when I quit a game first and am in the main menu. Always something!

TriForceX, 2020-09-27:
Well, you have to press for a few seconds, is not a direct button press, that just suspend the system, you need a long press. So you can confirm it just working with the start button?

Note: Yes the safe shutdown only work on main menu (you can also turn off from system menu > shutdown icon). Remember this is like a computer OS, so switching from physical button only is like removing the battery from a PC... Btw the "original firmware" is just an older version of our firmware, the "unclean shutdown" is still there and it can corrupt your system... they just removed the warning (a very bad thing)

MaciekBaron, 2020-09-30:
Thanks! That makes sense.

I can confirm that holding the start button brings up the power menu.

closed 2020-09-30 by MaciekBaron

creeper33, 2023-11-11:
hey umm.....i need help with my powkiddy Q90,so i dropped it and thankfully nothing bad happened.....except for the power slider to be broken, so i used gorila glue to glue a part of the slider back ((online it said that it was safe to use)), and it worked.....sorta, so when i would mess and wiggle the slider and weird ways, it would turn on, but when i let go or move it slightly, it turns off....i can send pictures if you need any :)

creeper33, 2023-11-11:
so is there like a way to turn it one without the slider or

nfriedly, 2023-11-13:
No, unfortunately, the slider switch is the only way to control power on these devices. Someone who does electronics repair could probably fix it, but it would probably cost more for their time than just buying a new one.

I'd recommend that you buy a replacement and then put the broken one on ebay "AS IS / for parts or repair". You can probably get $5-10 for it that way, and hopefully it will go to someone who can fix it on their own time.