TriForceX/MiyooCFW #108: Support L2 and R2 for PowKiddy V90

HoraK-FDF, 2020-09-29:
Hi,

is it possible to add support for the L2 and R2 Shoulder buttons of the PowKiddy V90?

TriForceX, 2020-09-29:
Yes, we are planning to make them available in the next release.

coldshouldermedia, 2021-01-30:
I'm looking forward to this, it'll make the ps1 experience awesome!

TriForceX, 2021-02-01:
The code already support L2 and R2, thats why this issue has been closed.
Is available if you build the source code, or just wait for the next release... i may release it as a patch, idk yet.

coldshouldermedia, 2021-02-01:
Thanks for responding, I chose to comment on a closed issue in an attempt to show excitement and support for the next revision, not seeking or demanding it.

I look forward to you implementing it!  Thanks for all your hard work!

BlueStarBRS, 2022-04-26:
@TriForceX  what is now with the l2 and r2 support ? it still dosnt work so what is going on ?

SnowKaya, 2022-05-07:
@TriForceX still waiting too, when we'll get a new release? Even if it's a minor one correcting this issues? At least a guide on how to manually build it from source

TriForceX, 2022-05-07:
> @TriForceX what is now with the l2 and r2 support ? it still dosnt work so what is going on ?

> @TriForceX still waiting too, when we'll get a new release? Even if it's a minor one correcting this issues? At least a guide on how to manually build it from source

There are no support from developers, only two guys started to help recently from time to time, but the process is slow due we have real life job and family, so any advance is done in free time. Remember this is a free open-source project, theres no estimated time for releases. 

About the L2/R2 the source code already has the support for it from many months ago, is a done task, but yea, we have not made a release due we have been working on leaving the source code stable and tidy, since previously for that reason there were no more updates, the code had been very messy and some codes were missing, we are still working on it, that is why we have not released a stable release .

Anyway the build guides are in the Wiki, and there are test packages in the MiyooCFW/sdcard repository, or getting an automated build from TriForceX/MiyooCFW in Actions tab. (Remember this test image are still in development).

As i have mentioned before in several places and to many people, we do not have enough support, there is no a team of developers behind this project, only 2 or 3 people who only help in their free time, I hope they understand the situation and be patient if you want an stable release.

SnowKaya, 2022-05-07:
Sorry, i had no idea, it's just that the long time since last release looks like the project has been abandoned or in it's best version. Please, take your time

ioma8, 2022-10-06:
This issue should not be closed as those buttons in any released version still do not work.

TriForceX, 2022-10-16:
> This issue should not be closed as those buttons in any released version still do not work.

theoretically an "issue" should be closed if it was solved on the project (source code), so developers doesnt have to care about it if someone already solved it... remember this is a project in development, released stuff is not usually related with current the progress of the project. issues from released stuff that have been solved are no longer an issue, keeping them opened will cause confusion for other developers.

about the releases i can understand that should be more often, but we cannot release stuff not properly tested, we did that in the past and it was very hard to maintain the project.

closed 2022-10-16 by TriForceX

BlueStarBRS, 2022-10-26:
> theoretically an "issue" should be closed if it was solved on the project (source code), so developers doesnt have to care about it if someone already solved it... remember this is a project in [...]

so l2 and r2 working now with a new released version ?