TriForceX/MiyooCFW #109: buildroots fails when patching package musl

HoraK-FDF, 2020-09-29:
Hi,

the following error happened to me can this somehow avoided?:

```
musl-1.1.24.tar.gz: OK (sha256: 1370c9a812b2cf2a7d92802510cca0058cc37e66a7bedd70051f0a34015022a3)
>>> musl 1.1.24 Extracting
gzip -d -c /STUFF/powkiddy/buildroot/dl/musl/musl-1.1.24.tar.gz | /STUFF/powkiddy/buildroot/output/host/bin/tar --strip-components=1 -C /STUFF/powkiddy/buildroot/output/build/musl-1.1.24   -xf -
>>> musl 1.1.24 Patching

Applying 0001-avoid-kernel-if_ether.h.patch using patch: 
patching file include/netinet/if_ether.h

Applying 0002-fix-risc64-conflict-with-kernel-headers.patch using patch: 
patching file arch/riscv64/bits/signal.h
Hunk #1 FAILED at 12.
1 out of 1 hunk FAILED -- saving rejects to file arch/riscv64/bits/signal.h.rej
```

gameblabla, 2020-10-01:
You can delete the patch 0002-fix-risc64-conflict-with-kernel-headers.patch in package/musl but normally it should build as of latest commit since i updated to GCC 10

HoraK-FDF, 2020-10-09:
I deleted the file "0002-fix-risc64-conflict-with-kernel-headers.patch" but then the next error appears I don't know what I can delete without breaking something:

```
musl-1.1.24.tar.gz: OK (sha256: 1370c9a812b2cf2a7d92802510cca0058cc37e66a7bedd70051f0a34015022a3)
>>> musl 1.1.24 Extracting
gzip -d -c /STUFF/powkiddy/buildroot/dl/musl/musl-1.1.24.tar.gz | /STUFF/powkiddy/buildroot/output/host/bin/tar --strip-components=1 -C /STUFF/powkiddy/buildroot/output/build/musl-1.1.24   -xf -
>>> musl 1.1.24 Patching

Applying 0001-avoid-kernel-if_ether.h.patch using patch: 
patching file include/netinet/if_ether.h

Applying 0003-nopic.patch using patch: 
patching file Makefile

Applying 0003-remove-riscv64-bits-user.h-contents.patch using patch: 
patching file arch/riscv64/bits/user.h
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file arch/riscv64/bits/user.h.rej
```

Does that problem has something to do with my machine here? Or does that happen to anyone?

gameblabla, 2020-10-09:
Well i guess it can be removed too but those errors are unusual... In fact those patches don't exist on my buildroot if you look at it here :
https://github.com/bittboy/buildroot/tree/bittboy_newish/package/musl

I think that the folder is still stuck on an older github branch instead of bittboy_newish.

TriForceX, 2020-10-09:
the miyooCFW repo is up-to-date with the bittboy_newish so is the same

https://github.com/MiyooCFW/buildroot/tree/master/package/musl

gameblabla, 2020-10-09:
Just look at the link you posted triforceX : it is on the bittboy_newish branch but it's an older commit...
https://github.com/bittboy/buildroot/commits/bittboy_newish

The "Fix extra quote." commit to be exact

TriForceX, 2020-10-09:
@HoraK-FDF @gameblabla true, my bad, i did a wrong merge from sauce's repository (bittbot_newish) so the files that was supposed to be deleted in the MiyooCFW repository never got deleted.

I fixed it show work now in https://github.com/MiyooCFW/buildroot

gameblabla, 2020-11-20:
That should be fixed now.

closed 2020-11-20 by gameblabla