TriForceX/MiyooCFW #111: Canonball: cannot change gear

cricri-pingouin, 2020-10-03:
There doesn't seem to be a button to change gear in Cannonball (Outrun emulator).
As a result, you are stuck in low gear, and don't stand a chance to reach even the first checkpoint.

djzipper, 2020-11-09:
It allows gear change when not pressing accelerate, but if you try and change gear at the same time nothing happens. It is currently unplayable. Currently unable to hold b and press 'R' to change gear at the same time.

cricri-pingouin, 2020-11-09:
Thanks djzipper!
I didn't get a reply so deleted the game from my V90 now, so can't test the gear changing while not accelerating.
Either way, this wouldn't be convenient seeing that the timer is quite tight already.

gameblabla, 2020-11-11:
I plan on fixing that at some point. Not sure how i screwed up that up but oh well.

yepyepyep4711, 2021-05-01:
Fixed with [this](https://github.com/gameblabla/Cannonball-C/pull/1), if it gets merged.