TriForceX/MiyooCFW #113: Atari800 Emulator

cmgonzalez, 2020-10-06:
To complete the 8bit Gang, we need an Atari800 emulator

Maybe someone can port this one...
https://boards.dingoonity.org/dingoo-technology-a380/dingux-atari-for-dingoo-a-380a-320e/

Regards

cmgonzalez, 2020-10-13:
This repo version almost compiles but at the end stop complaint about alsa_sdl

https://github.com/kerheol/dingux-atari

I just have to 
-edit makefile for bittboy sdk paths and remove -mips32 options
-comment function psp_sdl_load_png on sp2x_psp.c, to successfully compile

But I got this errors...
https://i.imgur.com/LLWr3Bh.png

gameblabla, 2020-10-13:
Add -lasound to LDFLAGS

nelosan01, 2020-11-09:
One Atari 800 emulator would be the best

salvacam, 2021-02-14:
I compile the Atari 800 and Atari 5200 emulator and dingux-atari. 
From here http://zx81.zx81.free.fr/serendipity/index.php?/categories/116-Atari-5200 

https://github.com/salvacam/dingux-atari/releases/tag/2021-02-13