TriForceX/MiyooCFW #114: Ngp & CPC slowdown

dredd666, 2020-10-08:
Im running cfw 1.3.3, overall everything os running smoother, comparing to prior ver. But new emulators ngp and Amstrad Run too slow, barely unplayable. Os This just me?

gameblabla, 2020-10-08:
This isn't you no. And if you were talking about NGPCEmu, well sometimes compiled a PocketGo/Bittboy of my NGPCEmu emulator but like i had to warn him that it was too slow for the hardware... Of course no one listens to me lol.
As for CrocoDS, i asked the main author of it for the source code and the github link to it but he has not updated his github repo yet.

dredd666, 2020-10-08:
Got it.
Thanks for replying

salvacam, 2021-01-20:
I have ported the GCW0 version of dingux-cap32. https://github.com/salvacam/dingux-cap32/releases/tag/2021-01-20