TriForceX/MiyooCFW #115: Image file contains empty partition

HoraK-FDF, 2020-10-09:
The file PowKiddy_Q90_V90_Only_CFW_v1.3.3_04-10-20.img contains empty space is this needed or is it simply a waste of space and bandwidth?

gparted reports that there is empty space of 1.5 GiB at the end and here is the output of 7z:
```
7-Zip [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
p7zip Version 16.02 (locale=de_DE.UTF-8,Utf16=on,HugeFiles=on,64 bits,4 CPUs Intel(R) Celeron(R) CPU J3455 @ 1.50GHz (506C9),ASM,AES-NI)

Scanning the drive for archives:
1 file, 4001366016 bytes (3816 MiB)

Listing archive: Powkiddy-V90-Q90_CFW_v1.3.3_04_10_20.img

--
Path = Powkiddy-V90-Q90_CFW_v1.3.3_04_10_20.img
Type = MBR
Physical Size = 4001366016

   Date      Time    Attr         Size   Compressed  Name
------------------- ----- ------------ ------------  ------------------------
                    .....    268435456    268435456  0.fat
                    .....    268435456    268435456  1.img
                    .....    268435456    268435456  2.img
                    .....   1580899840   1580899840  3.fat
                    .....   1614111232   1614111232  4
------------------- ----- ------------ ------------  ------------------------
                            4000317440   4000317440  5 files
```
I don't checked the other image files because I have no PocketGo.

salvacam, 2020-10-09:
You can add that empty space to the main partition, as explained in step 8 of the image
https://user-images.githubusercontent.com/16083854/59299539-e0ebb600-8c5a-11e9-91bb-f9386ec71d1c.png

HoraK-FDF, 2020-10-09:
Yes but this empty space in the file useless I mean it makes the download nearly double size for nothing. In version 1.3.2 there was not such an included empty space.

Apaczer, 2023-03-14:
There is no left-over space used in our latest images & there has been added auto ``resize`` option to fill unallocated disk space.