TriForceX/MiyooCFW #117: Hopefully the latest version of the FC emulator will be integrated into the mirror

ZOMBIE39, 2020-10-11:
Sorry, my English is not very good and I have to use a translation site to say what I want to say.I was very happy when I saw that the system was updated after more than a year.But I found out that the FC emulator is not the latest version from the author of the optimized version.The author of the optimized version of the emulator has since made another small update, the changelog mentions an increase in the speed of the emulator, the update date is October 23, 2019,I've posted the URL of the Baidu network drive below in the hope that your subsequent updates will integrate it.If you can't download it from the Baidu network drive, I can also upload it to other network drives.
https://pan.baidu.com/s/1TCCIfXfEPhYB3dI_1-DJ7Q
Extract the password:35x4
Finally, thank you again for not giving up on this system update yet.
[image]

TriForceX, 2020-10-19:
We will check this, thanks for sharing

andresito-dev, 2020-11-03:
I can't download from baidu. 
Can you upload it to another site?
Thank you very much.

andresito-dev, 2020-11-04:
> drives.

I can't download from baidu.
Can you upload it to another site?
Thank you very much.