TriForceX/MiyooCFW #118: Tv out

symbios24, 2020-10-13:
Hello is it possible for someone to tell me how can i add tv-out support on a bittboy v3.5 ?
I know some things from linux 
Thanks

gameblabla, 2020-10-13:
Hello, CVBS output from the headphone jack needs to be implemented in the kernel :
https://github.com/jamesofarrell/miyoo_kernel

> I know some things from linux

You would also need to know how to reverse engineer the soc because there's little documentation for it... and some C programming too... So yeah

symbios24, 2020-10-13:
Thanks,should i just add/implement somehow the working tv-out from the retrofw?

gameblabla, 2020-10-13:
No, the process isn't the same because said code isn't generic and very hardware specific.

symbios24, 2020-10-14:
So what is need is a genius to do it?

gameblabla, 2020-10-16:
> Or just take the code from the old image that worked in the bittboy v2?
What code ? If you were going to talk about the real time kernel, then it's not open source. Or at least, the bittboy specific portion of it isn't. So there is nothing to take if nothing was made public.
Also there isn't really any upstream code for Composite output on most allwinner chips but maybe it could be used for reference.

I hate to break this to you but kernel development isn't for noobs.

Apaczer, 2023-03-11:
fixed with: https://github.com/MiyooCFW/kernel/pull/43

closed 2023-03-11 by Apaczer