TriForceX/MiyooCFW #119: crocods Amstrad CPC emulator sound issue

reset-selectgame-1-9, 2020-10-15:
Is this a new port?
It has major sound and speed issues. Most of the games are impossible to play because they run too slowly.
Thanks for your amazing work!

gameblabla, 2020-10-16:
He just ported it because people asked and did not test for performance. He had been asked for the source code so we could take a look at it and see what's wrong with it but he didn't do so far so for now it will stay the way it is.

reset-selectgame-1-9, 2020-10-16:
thanks for the reply.
It's still awesome to have something to test than nothing. let's see how well this can be improved in future.

salvacam, 2021-01-20:
I have ported the GCW0 version of dingux-cap32.
You can download it from here https://github.com/salvacam/dingux-cap32/releases/tag/2021-01-20

G7200, 2021-01-20:
Amazing thanks, this one works much better!

closed 2021-10-15 by TriForceX