TriForceX/MiyooCFW #123: sdl2 needs a toolchain w/ dynamic library

HoraK-FDF, 2020-10-26:
I was trying to get the latest sdlpal (supports the english patched full versions for DOS/98) compiled but unluckily the toolchain does not have sdl2 so I tryed to configure buildroot with sdl2 but there are no packages available only a note "sdl2 needs a toolchain w/ dynamic library". Is it possible to somehow get the toolchain to one with a dynamic library+sdl2?

gameblabla, 2020-10-27:
I could try to add it based on that old fork :
https://github.com/pcercuei/SDL2
And some of those changes
https://github.com/od-contrib/buildroot-rg350-old-kernel/tree/1308a4c0ce87a8e5b8b8e29b261d8f748eb8ea3c/package/sdl2

But it's pointless in your case anyway because if you look closely, they still kept the SDL 1.2 rendering in video.c
https://github.com/sdlpal/sdlpal/blob/c0a00d190704e0a1171dedeb28e3b4d28b8ca6d1/video.c

Given how memory intensive and slow SDL2 is compared to SDL 1.2, i would look into that. Btw, the requirement for the dynamic library is just a check that can be removed. But it won't still work because SDL2 by default doesn't provide an fbdev backend unless you use pcercuei's fork. (and even then, it's an old version of SDL 2.)

HoraK-FDF, 2020-10-27:
I changed the Makefile a bit and now it compiles fine.
I changed the lines:
SDL_CONFIG = sdl2-config to SDL_CONFIG = /opt/bittboy-toolchain/arm-buildroot-linux-musleabi/sysroot/usr/bin/sdl-config
LDFLAGS += -lGL -pthread to LDFLAGS += -pthread

The game starts now but for the dos version I have shuttering sound and for the win98 version there plays no midi, also there is an cursor visible in the dos version.

here is the contents of the ini file maybe you can see what needs to be done if it can be done by simply editing the config file to get the sound running:

--- sdlpal.cfg start ---
#####################################################
# Configuration file for SDLPal WIN game files      #
#                                                   #
# Lines started with '#' is treated as comments.    #
#####################################################

# KeepAspectRatio: Indicates whether the content in game window should keep its original
#                  aspect ratio (default) or not. When aspect ratio is kept, there may be
#                  left/right or top/bottom black edges in the game window. Currently this
#                  value is only valid if the game is built with SDL 2.0 or higher, otherwise
#                  this value is ignored.
KeepAspectRatio=1

# FullScreen: Indicates whether the game should started fullscreen or not, default is not.
#             On some platforms, such as mobile OS or WinRT, this option is ignored.
FullScreen=1

# LaunchSetting: Launches setting interface on next game start, 0 (default)
#                1 for launch and ALL comments will be deleted!
LaunchSetting=0

# Stereo: Indicates whether to use mono audio or stereo audio. Non-zero means to use
#         stereo audio (default), while zero means only use mono audio.
Stereo=1

# UseSurroundOPL: Indicates whether to use surround opl emulator or not. Non-zero
#                 means to using (default), while zero means not to use. Only
#                 valid when Stereo is set to non-zero.
UseSurroundOPL=1

# EnableKeyRepeat: Allows continuous input when key is held down.
#                  0 for disable (default).
EnableKeyRepeat=0

# UseTouchOverlay: Translucent button overlay on game screen for touch-based platforms.
#                  0 for disable (default).
UseTouchOverlay=0

# EnableAviPlay: Play AVI video clips, only for windows game files.
#                0 for disable (default).
EnableAviPlay=1

# EnableGLSL: Recently a EXPERIMENTAL hacked OpenGL/GLES based backend was introduced, which
#             accepts RetroArch GLSL Shaders. This option MUST be used with Shader option.
EnableGLSL=0

# Shader: Path to RetroArch GLSL Shaders( see https://github.com/libretro/glsl-shaders ).
#         Either glsl or glslp; either absolute path or relative to GamePath
#Shader=/Applications/RetroArch.app/Contents/Resources/shaders/shaders_glsl/xbrz/xbrz-freescale.glslp

# EnableHDR: ACESFilm based HDR effect. Only valid when EnableGLSL=1.
EnableHDR=0

# TextureWidth/TextureHeight:
#   1. When EnableGLSL=1, the final(before blitting) texture size. If your mobile device have very
#      high resolution and the GPU cannot afford the Shader specified, try set a lower texture size.
#   2. When EnableGLSL=0, only specify the ratio that KeepAspectRatio use.
TextureHeight=400
TextureWidth=640

[... 69 lines cut ...]
# OPLCore: Indicates the OPL emulator core to use. Valid types include 'DBFLT'
#          (default), 'MAME', 'DBINT' and 'NUKED'.
OPLCore=DBFLT

# OPLChip: Indicates the kind of OPL chip to be emulated. Valid types include 'OPL2'
#          (default) and 'OPL3'. Note: the 'NUKED' core does not support 'OPL2'.
OPLChip=OPL2

# MessageFileName: Indicates whether to use external message files.
MessageFileName=m_eng.slf

# FontFileName: Indicates whether to use external font files. Currently supports only BDF font.
#FontFileName=jm.msg

# LogFileName: When set, the program's log will be output to this file as well.
#LogFileName=sdlpal.log

# GamePath: File directory for game files, './' (default) uses the same directory as
#           this config file
GamePath=./

# SavePath: File directory for save files, './' (default) uses the same directory as
#           this config file
SavePath=./
--- sdlpal.cfg end ---

gameblabla, 2020-10-27:
It sounds like the sound is messed up simply because it can't keep up. It would try to change the resolution to 320x240 and disabling stuff like UseSurroundOPL=1 or change stuff like setting ResampleQuality=4 to ResampleQuality=0.
The AudioBufferSize=512 setting seems inappropriate too so try increasing it to AudioBufferSize=1024 or AudioBufferSize=2048.

Also, MIDI playback on this handled is CPU intensive. Use OGG or MP3 instead.
So set Music=MIDI to Music=OGG or Music=MP3. (assuming you have ogg or mp3 files for those)

Cheers

gameblabla, 2020-11-20:
Let me know if what i told you improved performance. SDL2 won't get ported to this handled due to memory contraints and higher CPU overhead.

closed 2020-11-20 by gameblabla

HoraK-FDF, 2020-11-24:
I now got the sound working right feels fluid and good timed. Now I fiddling a bit with the scaling. Id like to remap the keys a bit but at the moment I don't know where to look for that maybe you have a hint for me? When I all got working as best as I can get it should I upload the whole packages here? The DOS version is abandon ware and the Windows 98 version is Freeware.