TriForceX/MiyooCFW #124: The PocketGo intro is still shown in front of the new PowKiddy intro

HoraK-FDF, 2020-10-26:
The PocketGo intro is still shown in front of the new PowKiddy intro, I don't know if this is intended this way but it looks odd so I thought it may be an issue.
used version: PowKiddy_Q90_V90_Only_CFW_v1.3.3_04-10-20.img

gameblabla, 2020-10-27:
Because i believe that one is the bootloader and the other one is an executable that gets executed after. Is there a boot-logo executable on your FAT32 partition ? you can remove it.

HoraK-FDF, 2020-10-27:
ok the one with PocketGo is the bootloader and the PowKiddy one is the boot-logo. Is it possible to disable the bootloader intro? Since I'm here on a PowKiddy V90 the PowKiddy intro would fit here better..

gameblabla, 2020-10-27:
For the bootloader, you can recompile it to remove the logo or just display the kernel loading up buildroot but unlike boot-logo, it would be pointless to do because it would barely the boot times. (if at all) Recompiling it is not a simple process.

closed 2023-03-14 by Apaczer