TriForceX/MiyooCFW #125: Redo the rootfs/make the CFW less of a mess

gameblabla, 2020-11-13:
So far, everything was prebuilt & assembled manually. Obviously this is problematic as it prevents us from maintaining properly.
The correct way of doing things would be like Opendingux does : put everything (except apps/games/emulators) in buildroot.
This means that daemon programs should also be built separately as separate programs.

I have a WIP rootfs here : https://github.com/bittboy/rootfs
Right now, it uses glibc but final version will be most likely using musl libc unless there are other reasons. (i'm thinking of midpath)
If CedarXPlayer is working though, then i may stick with glibc. (Naturally, it wouldn't be ideal)

- [X] Working and bootable rootfs, complete with a working kernel built from buildroot.

- [ ] Make CedarX work for hardware video acceleration. (See https://github.com/gameblabla/kernel)

- [ ] Gmenu2x or GmenuNx with OPK container support for ARMv5, brightness control and CPU overclocking.

- [ ] Add bittboy specific programs like daemon, miyooctl to the list of buildroot package.

- [ ] Fix building issues with uboot/buildroot, so a complete sd card can be built from buildroot.

I cannot consider it complete unless all of the tasks are done. (except for maybe the last one since it's optional but it would be extra nice, naturally)

gameblabla, 2020-11-16:
So i got CedarX to mount itself in /dev/ but any attempt at making software use it has been unsucessful so far...
I'm not sure if it was even tested. And then there's the issue that CMA does allocate memory from the bittboy. Now, it is possible to reduce the kernel memory usage (up to 1MB so far) to some extent as to mitigate the increased memory usage but even if i do get it to work, it's something i have to think about and whenever it is worth it...

gameblabla, 2020-11-20:
So aodzip will look into the F1C100S Cedar issue, now i'm waiting back from him...
Other than that, no other progress had been made. I am considering to use upstream kernel with patches on top given that they support the screen for the PocketGo/3.5...

Apaczer, 2023-04-17:
Can you tell what will be a practical use of CedarX for Bittboy platform?

We have functional uClibc & musl image build from "scratch" at https://github.com/MiyooCFW/buildroot.

The brightness control and CPU overclocking for GMenu2X is already a thing in [source](https://github.com/MiyooCFW/gmenu2x). OPK is working but planned to be implemented in CFW probably later.

Shortcuts placement can be controled partly with ``miyooctl -k`` and have been moved from daemon to kernel by last CFW's developer I would rather not redo smth that has been intentionally undone.

szymor, 2023-04-17:
@Apaczer lower CPU usage at audio and video playback, possibility to play videos encoded with real-life codecs
In general, all current apps and games could benefit from it if they play any audio, though I am not sure whether it would be impactful.

gameblabla, 2023-04-18:
Apparently some had gotten it to work here but it was "less than optimal" : https://whycan.com/t_4219.html.
I suppose if OPK support is released, we could close this.

Apaczer, 2023-07-15:
added an entry to ``/mnt/options.cfg`` to optionize shortcut placement by enduser:
`HOTKEY_CUSTOM=0` -> default kernel hotkeys
`HOTKEY_CUSTOM=1` -> use daemon shortcuts from `/mnt/.buttons.conf`

Apaczer, 2023-12-19:
- CedarX not usable under MiyooCFW with F1C100S soc, see for ref.: https://github.com/MiyooCFW/kernel/pull/78#issuecomment-1838093650 (crossing that one out of the list as a result)

- For now added IPK support in GMenu2X and OPKG in BR2 cfg for obvious reasons.

Apaczer, 2026-09-28:
we got Cedar working for decoding (to less CMA resources for encoding) a while ago in musl (no release of uclibc variant for libcedarc.so by Allwiner), so closing as all relevant points have been resolved. For those interested there is WIP implementation of zero-copy hw playback in our BR2 repo.

closed 2026-09-28 by Apaczer