TriForceX/MiyooCFW #129: [Feature Request] Spectrum fuse compatibility maps

nportillo2, 2020-11-24:
Spectrum controls are really a mess, but I hace about 700 maped games, just start and play like a arcade.

You can map games from the menu, but you can't save a keyboard mapping for each game, so you have to start procedure for each game.

Should it be possible to save config for each game? Also... use fuse emulator format?

Also disable Border would be a good feature, as screen is tiny.

salvacam, 2020-11-24:
The eumlador is the Unreal Speccy Portable, for other similar consoles they have carried the Fuse, it has not yet been possible to port to the bittboy/pocketgo

There is a new version [https://github.com/salvacam/speccy/releases](https://github.com/salvacam/speccy/releases) that saves the control mapping settings, the joystick type and whether or not to use fullscreen.

https://www.youtube.com/watch?v=cxj3KcN1ZEc&ab_channel=Salvacam

You can update this emulator. Replace the file in the "main" partition "emus/speccy/pocketSpeccy" with this file https://github.com/salvacam/speccy/releases/download/0.1.9/pocketSpeccy
It is recommended to rename the current file, in case of error

salvacam, 2020-11-28:
> Spectrum controls are really a mess, but I hace about 700 maped games, just start and play like a arcade.
>
> You can map games from the menu, but you can't save a keyboard mapping for each game, so [...]

There is already a version of the Fuse emulator for Miyoo devices, please note that it is a preview version and may be buggy.
https://github.com/salvacam/fuse-od/releases/tag/0.0.1Miyoo

closed 2021-03-10 by TriForceX