TriForceX/MiyooCFW #131: how to make pocketgo's usb-otg work

oomhj, 2020-12-02:
how can i make pocketgo's usb-otg work,so we can make connect external usb device possible。like  wireless card or U disk

RichardNobel, 2020-12-02:
That would be nice. 🤓 _(I have a PowKiddy V90, a "Clamshell PocketGo")._
But, over [here](http://wagnerstechtalk.com/pg2/) I found this observation: 

> **USB Devices with an OTG Adapter (USB-C to USB-A) do not appear to work** – At this point, it does not appear that additional devices can be used with the PocketGo v2. I tried an OTG Adapter with the WiFi Adapter (used on my RG-350) with GCW Connect and had no success with the PocketGo v2. I also tried a folding USB keyboard with no success. Until I find out otherwise, I’ll have to conclude these type of devices do not work with the PocketGo v2 at this time.

Haven't tried it myself, yet. But I bet the included USB (type C) cable is only for charging, with data pins/lines missing. _(In general I've never understood why leaving out _just a couple of pins/lines_ would reduce cable manufacture costs that much... oh well 😕 that's a different topic)._

oomhj, 2020-12-02:
I disassembled the device and found that OTG circuit was missing and USB-D1 was directly grounded。So, i will try set the usb-driver to HOST mode. And use an externally powered USB-hub

oomhj, 2020-12-08:
pocketgo's SDIO1 is multiplexed with SPI0. But only in 1bit mode. can it be used for wifi module?