TriForceX/MiyooCFW #134: Is there any way to support exfat or ext3/4 for the main partition?

sentisk, 2021-02-02:
The main partition is fat32 now, as we know, which is not good for files bigger than 4GB. So is there any way to support exfat or ext4 like retropie? Thanks.

xXS4ALLx, 2021-03-22:
What files are you putting on the sd card that is bigger than 4gb?

Apaczer, 2023-12-20:
To acces ext4 you've to enable EXT4_FS config on [kernel](https://github.com/MiyooCFW/kernel) (we have removed support for ext4 in recent commits), then use [buildroot](https://github.com/MiyooCFW/buildroot) and edit your BR2 configuration by pointing to your custom linux kernel. MAIN needs lastly to have partition type corrected in [genimage-sdcard.cfg](https://github.com/MiyooCFW/buildroot/blob/130d2be1f31779b3f3a36cdaa0bc69086a3e1c16/board/miyoo/genimage-sdcard.cfg#L51C1-L55) and customize auto-gen of `main.img` file in [genimage.sh](https://github.com/MiyooCFW/buildroot/blob/130d2be1f31779b3f3a36cdaa0bc69086a3e1c16/board/miyoo/scripts/genimage.sh#L50) script. For ROOTFS filesystem change you must modify also BR2_TARGET_ROOTFS in "Filesystem images" options. 

Anyway we switched to BTRFS for both `rootfs` and `main` which should solve your problem.

closed 2023-12-20 by Apaczer