TriForceX/MiyooCFW #137: Stuck on Start Intro

Carnagehunts, 2021-02-10:
I've seen the issue at other moments say that this is caused by either improper shutdown or clicking all 4 buttons at the same time and yes I've had this happen to me. In this case I had properly closed pocketgo and went to add some games and add cht files, SAFELY eject from pc and when powering on the pocketgo it no longer accepts the sd and stays stuck on intro. Then when going to check on my pc if something happened the pc immediately says that the drive is not accepted and to please format drive. This of coarse shows the drive as corrupted and I'm once again forced to redo the entire process of flashing the drive and partitioning the card. Has anyone else seen this and could this be a SD Card related issue or based on the 1.3.3 update? I was using Samsung 64gb Evo sd card for all the games. I have repeated the process 3 times just today and each time had to reflash due to pocketgo just outright get stuck on intro.

TriForceX, 2021-02-18:
> I've seen the issue at other moments say that this is caused by either improper shutdown or clicking all 4 buttons at the same time and yes I've had this happen to me. 

You need always to do a proper shutdown from software and then form hardware (physical button). The instructions are in the main wiki page [here](https://github.com/TriForceX/MiyooCFW/wiki#shutting-down-the-console)

> Then when going to check on my pc if something happened the pc immediately says that the drive is not accepted and to please format drive. This of coarse shows the drive as corrupted and I'm once again forced to redo the entire process of flashing the drive and partitioning the card.

I assume youre using windows, so windows cant read linux filesystem by default, so 2 of the 4 partitions are not loaded by windows and marked as unformatted drive (the sd card always contains 4 partitions)... The only 2 partitions working are "boot" and "main", and the "main" one is where the general cfw stuff is.

> Has anyone else seen this and could this be a SD Card related issue or based on the 1.3.3 update? I was using Samsung 64gb Evo sd card for all the games. I have repeated the process 3 times just today and each time had to reflash due to pocketgo just outright get stuck on intro.

Are you formatting with the recommended tools (SD Card Formatter for example) before install the cfw. Are you extending the "main" partition before use for the first time in the console?

Carnagehunts, 2021-02-18:
I've done the formating, extending all of it I have around 40+gb of games in the sd card it's only possible with extending drive. That's why my main question is if the problem could be related to the type of card used or if maybe something is getting corrupted during the process of adding to the sd card. I repeated the process 3 times and have closed pocketgo the proper way each time and after adding something new to the drive like a cht file or new game. The pocketgo won't launch and stay stuck in intro screen. After removing card the drive is instantly corrupted and no longer runs so must be reformated again.

xXS4ALLx, 2021-03-23:
Just throwing some ideas here, you could be a fake sd card? If the true size of the sd card is 16gb and you copy over this size, the card gets corrupted.
Only way to find out if your card is fake or not is use a popular app called "h2testw", google it, download it and test your sd card, then report back here.

closed 2023-03-14 by Apaczer