TriForceX/MiyooCFW #138: Hey TriForceX!

BryMD, 2021-02-19:
Hey @TriForceX !

I have a wad of ready-made improvements to the CFW I need to get off my back. As a medical doctor in the middle of a global pandemic I just can't be bothered with learning Discord, so making it easy for myself and dumping it here :P

A quick rundown on my background: Have passionately been following the development of GMenu2X ever since the GP2X days, but always found a better reason not to invest time or money in an actual console (read: a Windows netbook, my first iPhone, my very own Mac-museum). During the pandemic I've needed a portable project to keep me sane between my shifts, and the PocketGo simply came at the right time at the right price.

Then on to the (subjective) improvements I've made:

NEW ICONS
In my opinion FontiGrid is still the most unified and complete theme for GMenu2X, though still not perfect. As such, I've designed new icons for the following:
1. UAE4All (pixel art: Amiga 500 Competition Pro joystick, background: Mr. Nutz Hopping Mad)
2. J2ME (pixel art: Nokia 3310, background: Knight Tales Land of Bitterness)
3. ScummVM (pixel art: 1986 Apple mouse, background: Monkey Island 2 LeChuck's Revenge)
4. Just4Qix (2 versions based on Chihun's amazing works - I have NOT requested approval from the artist himself)
New icons for FinalBurn Alpha, DOSBox and szymor's awesome YATKA are in the works.

NEW SHORTCUTS
I'm completely blank on Linux-programming, but have found the following games working on existing ports:
1. The Chocolate DOOM-port fully supports the commercial total conversion HacX: Twitch 'n Kill (requires hacx.wad)
2. The Hacked Hexen-port fully supports the expansion pack Hexen: Deathkings of the Dark Citadel (wonky background intro at the title screen but otherwise perfectly playable, requires hexdd.wad)

NEW SORTING
No one has bought a PocketGo to wordprocess in WizWrite, so why on earth is Applications still the first section (other than simply maintaining alphabetical order)?! Some shortcuts default to folders that MUST be named 'applications' and 'settings', so have myself concocted a down and dirty workaround by enumerating the remaining folders as '1. emulators' and '2. games'. This will sort the important sections first and the enumeration will remain hidden in their titles.

I'm attaching the improvements as a ZIP-file to this post. Use the improvements or don't. Regardless, this is my current contribution to the community :)

Keep up the AWESOME work!

Sincerely,
BryMD

[brymd.zip](https://github.com/TriForceX/MiyooCFW/files/6013015/brymd.zip)

BryMD, 2021-02-19:
And for everyone else: Here are the icons:

[image]
[image]
[image]
[image]

BryMD, 2021-03-14:
Hi Guys!

Appending additional improvements:

NEW GAMES
1. The Quake-port supports the official Mission Pack 1: Scourge of Armagon (requires the mission pack's PAK-files in a folder named 'hipnotic').
2. The Quake-port supports the official Mission Pack 2: Dissolution of Eternity (requires the mission pack's PAK-files in a folder named 'rogue').

NEW ICONS
1. UAE4All (update: removed regrettable antialiasing on the pixel art)
2. YATKA
3. Quake (subjectively better icon)
4. Quake: Scourge of Armagaon
5. Quake: Dissolution of Eternity

As always, attaching the improvements as a ZIP-file to this post.

Grind on!

Sincerely,
BryMD

[brymd2.zip](https://github.com/TriForceX/MiyooCFW/files/6137514/brymd2.zip)

TriForceX, 2021-03-21:
Thanks for this updates, we have this pinned on the main to-do list for the next release.

BryMD, 2021-03-22:
No worries, @TriForceX !

PS: If possible, keep this issue open! I've got more stuff on the way ;)

Sincerely,
BryMD

BryMD, 2021-04-05:
OK, so here is my final batch for now.

NEW GAMES
The following supported games running on the Chocolate Doom-engine are quite obvious, though not always as easily remembered. As such, I have created separate shortcuts for each separate of the following games. Furthermore, the shortcuts more appropriately now look for their respective WADs inside the 'ccdoom' folder in 'games' instead of inside 'pc' in 'roms'.

1. The Chocolate Doom-port supports Doom II: Hell on Earth (requires doom2.wad)
2. The Chocolate Doom-port supports the official Mission Pack 1: TNT Evilution (requires tnt.wad)
3. The Chocolate Doom-port supports the official Mission Pack 2: The Plutonia Experiment (requires plutonia.wad)

NEW ICONS
In respect to the latter 2 icons, the already existing icons have been repurposed for Mission Pack 1 and 2.

1. DOSBox (background: Zool)
2. Doom
3. Doom II: Hell on Earth

DOS
We are all aware that the DOSBox-port is painfully slow. Even getting LaunchBox 2.0 to load takes ages. LaunchBox 1.1 is quite a lot snappier, though at the expense of categories. Furthermore, I have found the following VGA games to be (barely) tolerable.

1. Zool
2. Stone Age
3. The Adventures of Willie Beamish

As always, attaching the cumulative improvements as a ZIP-file to this post.

I'm now moving on to testing game-compatibility of @pthalin 's awesome J2ME-port, as well as the Custom PCSX-Rearmed Config Library.

Keep up the good work!

Sincerely,
BryMD

[brymd3.zip](https://github.com/TriForceX/MiyooCFW/files/6260352/brymd3.zip)

TriForceX, 2021-04-08:
Nice updates, do you think you can make the zip package as an standalone file? i mean having the same structure as the gmenu and its skins stuff? if not dont worry, we can do it later, but if you have some time to do it would be appreciated.

BryMD, 2021-05-25:
Hi @TriForceX !

Man, been battling myself for the past couple of months in regards to your request, and sadly I cannot comply with arranging all my updates in the CFW folder structure for the time being. As a medical doctor in the middle of a pandemic, there simply isn't enough time available for me between shifts. Sorry about that!

That said, here are my freshest contributions to the CFW:

NEW SHORTCUTS
1. Added 'selectorfilter=.gba' to the gpSP GBA emulator shortcut to hide .sav-files.

NEW ICONS
1. Brand new custom WizWrite icon aligning its visual impression with the remaining application icons.

NEW SKINS
1. Fixed color map of the MAME4All skin.
2. Updated the WizWrite skin for a more modern (and less skeuomorphic) look.
3. Changed the Zelda Picross intro graphics for a more professional presentation.

I still have a crap-ton (metric) of additional improvements on the wing, but haven't found time to structure them properly for upload yet.

Attaching the cumulative improvements as a ZIP-file to this post.

Sincerely,
BryMD

[brymd4.zip](https://github.com/TriForceX/MiyooCFW/files/6539594/brymd4.zip)

BryMD, 2021-05-30:
Just a quickie update:

NEW ICONS
1. Sqrxz
2. Sqrxz 2
3. Sqrxz 3
4. Sqrxz 4

PS: For those who didn't know, Szymor has been awesome again and ported Sqrxz 1 as well as Xump to the BittBoy/PocketGo as well!

Updates added to the cumulative ZIP-file.

Sincerely,
BryMD

[brymd5.zip](https://github.com/TriForceX/MiyooCFW/files/6566210/brymd5.zip)

BryMD, 2021-09-13:
Hi guys!

OK, so here are my progresses as of recent:

NEW ICONS
1. New (and subjectively better) Pico-8 icon (pixel art: Lexaloffle's own imagining of the Pico-8 gamepad, background: Celeste)

NEW SHELL SCRIPTS
1. scummlaunch. I've created a simple script for the ScummVM emulator that enables launching games directly from GMenu2X. It requires some manual labor though, as each user first has to add all their games to the default ScummVM interface, and then has to make text files for all their games containing the 'game short name' (found on the 'compatibility'-page of the ScummVM website) and having a .scumm filename suffix. For instance, the 'Monkey Island 2: LeChuck's Revenge' text file can be named 'Monkey Island 2 [LeChuck's Revenge].scumm' and contain the game short name 'monkey2'. I'll make a 'Read Me' with proper instructions when time suffices...

NEW SHORTCUTS
1. FAKE-08 Pico-8 emulator. I've found FAKE-08 to both have superior game compatibility, better graphics compatibility, and better sound compatibility. Furthermore FAKE-08 indeed supports launching games directly from GMenu2X!
2. mplayer application. The run.sh is quite the ingeniously written script for resuming movies. However, it has a terminal flaw that once the movie has played once it cannot be played again. To add insult to injury, it spams the movies folder with hidden resume-files. The new shortcut simply calls on the application instead of the shell script, thus bypassing it.

Now back to the trenches :)

Sincerely,
BryMD

_PS: I've radically cleaned up file tree of my install of the MiyooCFW to better mirror what is presented in GMenu2X. I've tried my best to restore all the attached shortcuts back to default paths, but no promises! As such, I'd recommend the attached shortcuts to be beta-tested before going live ;)_

[brymd6.zip](https://github.com/TriForceX/MiyooCFW/files/7157495/brymd6.zip)

BryMD, 2021-09-16:
Already prepped a new update:

NEW ICONS
1. UAE4All (update: cleaned up pixel art)
2. Sega 32X (background: Knuckles Chaotix)
3. MAME4All (background: Hyper Pac-Man)
4. FinalBurn Alpha (background: Metal Slug)
5. Snes9X4D (fixed vertical alignment)
6. SMS (duplicated SMS_SDL icon)

NEW SHORTCUTS
1. SCUMM. New shortcut calling on my 'scummlaunch' script instead of ScummVM itself.
2. DOS. DOSBox surprisingly also supports launching games directly from GMenu2X by default(!). This requires each user to create separate '.conf'-files for all of their games containing the required commands to launch them in the '[autoexec]'-section of the '.conf'-files. However, this requires no more manual labor than adding these commands to LaunchBox, and, as such, renders LaunchBox superfluous.

Enjoy!

Sincerely,
BryMD

[brymd7.zip](https://github.com/TriForceX/MiyooCFW/files/7181008/brymd7.zip)

TriForceX, 2021-09-16:
@BryMD many thanks! for now another group with me is working on testing the next release, we are fixing few things, then we will include all of this stuff. 

Im still following your updates, thank you very much

BryMD, 2021-09-17:
Hi @TriForceX !

No worries, man! After all, you guys are doing all the important heavy lifting around here ;) I'm simply polishing the surface you have provided with the CFW to give it the most logical, most consistent, most integrated, most aesthetically unified, and easiest-to-use experience the CFW can offer. And I'm quite aware that none of these hyperboles are particularly important to the CFW's target demographic - but they are very important to me personally, and I'm simply dumping all my progresses towards this goal here in this thread just in case others find them valuable as well ;)

Keep up the awesome work, you guys!

Sincerely,
BryMD

BryMD, 2021-09-28:
Another quickie update:

NEW ICONS
1. HacX

NEW APPLICATIONS
2. Bard. The Bard application (at least the one in 1.3.2) is broken. I dug up an update for Bard by Reddit user AgvirtheSilent ([https://www.reddit.com/r/Bittboy/comments/eci9kk/bard_with_gcw0_button_support/](https://www.reddit.com/r/Bittboy/comments/eci9kk/bard_with_gcw0_button_support/)) that works (albeit barely).

Sincerely,
BryMD

[brymd8.zip](https://github.com/TriForceX/MiyooCFW/files/7243775/brymd8.zip)

BryMD, 2021-10-05:
Final update for now:

NEW ICONS:
1. L'Abbaye des Morts
2. GMU (background: iTunes Visualizer/G-Force)
3. ST (fixed alignment of icon)

Enjoy!

Sincerely,
BryMD

[brymd9.zip](https://github.com/TriForceX/MiyooCFW/files/7288875/brymd9.zip)

closed 2021-10-15 by TriForceX