TriForceX/MiyooCFW #140: PS1 not running on V90

Nochobrine, 2021-03-01:
I am not sure what the issue is, but Ps1 games refuse to run on my V90.

When I select a file to launch (.cue or .pdp), the emulator launches but the game doesn't start. It goes directly to the in-emulator settings.

I have tried using HLE and SCPH1001, but neither make a difference.

I have tried both the version of pcsx that is included with the .img, and the update suggested by Retro Game Corps in his review ( https://retrogamecorps.com/2021/02/01/review-powkiddy-v90/ ). It seemed to make no difference.

I have tried reformatting the sd card multiple times. I even tried two other cards. The only difference that made was that it worked directly after reinserting the reformatted card to test if it worked. It would work fine. However, after I would take it out (using system shutdown and not just external switch) to add the rest of my roms back, it would no longer work.

TriForceX, 2021-03-21:
are you using the included sd card?
are you using the included firmware?

Nochobrine, 2021-03-21:
I am not using either the SD card or software that came with the V90 when I first got it. I am currently using the Miyoo firmware on a PNY 16 GB card. I have tried other cards however (excluding the one that came with the V90 because I know they tend to be sketchy), and it made no difference.

xXS4ALLx, 2021-03-22:
Can you press SELECT then EDIT on the PS1 emulator and check the CPU clock speed?

It could be the CPU clock speed that is currently set is unstable, try changing to 732 and press START to save the settings and try booting the PS1 game again, this also applies to other emulators as well.

Nochobrine, 2021-03-22:
Setting both the PS1 and GBA clock speed to 732 didn't seem to to anything.

xXS4ALLx, 2021-03-22:
Interesting issue this is, do you have another card reader that you could use? Also what type of card reader you using?
I have a usb3.0 multi card reader, when use this, it randomly corrupts files that I copy over, so now I just use the built in one that is on my laptop, files copy slower but not a big deal as long as the files are not getting corrupted.

Nochobrine, 2021-03-22:
The card reader I am using is "USB Type -C/OTG Card Reader/Writer" that came from Walmart. I've been using it for almost a year now and had not had an issue with it before. I don't have another one I could use.

xXS4ALLx, 2021-03-23:
Reading through all your posts to get a clearer picture of your scenario and it looks as though you pretty much tried what I would have tried besides using another card reader. I'm not saying that this might be the cause of your issue or not but it's something that you have not tried, maybe try to get a normal USB card reader (not usb-c) even though you used this in the past and it works, maybe there is something that the v90 doesn't like when you start copying files over. Since you said everything works fine the first time the v90 uses the sd card, and you start having issues when you start copying roms to the sd card.
Worst case scenario you have faulty unit?

Nochobrine, 2021-03-23:
To clarify, it only worked when first inserting the sd card once. Now every time I try, it just doesn't work. I just grabbed my old Macbook Air running Windows, and I'm going to see if using that could fix something. I'm going to try with both the main card I've been using (PNY 16 GB) and a Sandisk just to double check.

Also just to make sure that this isn't an issue, when I go to put my roms and skins back on, I copy the folders from :main that I had changed (emus, gmenue2x, roms ONLY) in their entirety and paste them back onto the card.

xXS4ALLx, 2021-03-23:
You using Mac or Windows?

Nochobrine, 2021-03-24:
I am using Windows on both my main PC and the laptop I mentioned.

The sd card that came with the V90 already had PS1 games on it when I got it. The games on there seem to run fine.

The issue seemed to just happen after I would load the SD card on the V90. It was only the one time when I got it to work somehow. I have yet to get it to work again. I tried loading both sd cards on the laptop with no luck. I am going to mess around with a couple small things here and there to see if I can make something happen.

xXS4ALLx, 2021-03-24:
Just covering all bases here, so uou are flashing the v90 CFW v1.3.3 to the sd card correct from this GitHub?

Also, once the sd card has been flashed, you insert the sd card back into your v90, everything boots fine?

Now when you connect the sd card to your PC to add roms, are you adding ROMS only and nothing else?

Once your done copying your roms, you insert the sd cRd back into your v90 and thats when issues start happening?

Nochobrine, 2021-03-24:
You are right apart from roms being the only thing I added. I would normally add some custom skins I have, along with a couple emulator updates that were suggested here (https://retrogamecorps.com/2021/02/01/review-powkiddy-v90/). I haven't done the emulator updates for a while now because I was just trying to get the PS1 to run. 

I did end up getting the PS1 emulator to work. I am not sure what I did differently this time, but it seems to be working ok for now at the stock clock speed after re-imaging the card. GBA still seems to have the issue I talked about earlier.

xXS4ALLx, 2021-03-24:
Ok, that's a start, the GBA games that you still have issues with, is the GBA emulator CPU Clock speed is also on stock 702?

Also, are you using the official GBA bios? I know some games work better with GBA Bios, the GBA emulator has it's own built in bios, that too also works better on some games, so it's a matter of trial and error.

Nochobrine, 2021-03-25:
The clock speed is set to 702. I'm not sure what you mean specifically by the "official bios", but it is the same one that I got from the suggested emulator updates.

closed 2021-10-15 by TriForceX