TriForceX/MiyooCFW #141: System Date/Time reset after restart

feurigel, 2021-03-14:
I set the date and time to current values.
If I restart the device (shutdown + power on), the date and time have been reset.

This worked with default firmware.

Device: Pocket Go v1
CFW: 1.3.3

TriForceX, 2021-03-15:
what you mean with "reset"? because no one of the miyoo consoles (bittboy, pocketgo, powkiddy) doesnt support real-time clock, so date and time is not updated...

feurigel, 2021-03-15:
Hi,

sorry, my description was a bit misleading and confusing.

Steps:

- I set date and clock to: 2021/03/15, 18:50
- Save with pressing "Start"
- Switch off Pocket Go
- Switch on Pocket Go
- Date and clock were reseted to: 2019/08/19, 17:09

If the systems don't have realtime clock, then we don't need this function of course. I thought it was integrated because of this setting. 😃

Cerezk, 2021-12-01:
> what you mean with "reset"? reset to what? because no one of the miyoo consoles (bittboy, pocketgo, powkiddy) doesnt support real-time clock, so date and time is not updated...

Oh, so none of these devices actually have an internal hardware clock? :O (I had no idea!)

Issue #245 is essentially a duplicate/expansion on this. They could be merged:
https://github.com/TriForceX/MiyooCFW/issues/245

Apaczer, 2023-03-14:
fixed from gmenu2x end on our src, but may be added globally in e.g. ``main`` script later to work with other frontends from the gecko. Keeping https://github.com/TriForceX/MiyooCFW/issues/365 as "enhancement" notice.

closed 2023-03-14 by Apaczer