TriForceX/MiyooCFW #143: gpSP image on Powkiddy V90 is blurry in every display scaling option but "unscaled 3:2"

AndrewOrso, 2021-03-16:
Using all type of display scaling but "unscaled 3:2" will result in a very blurry image (with illegible texts) using gpSP. The screen filtering option seems not working at all (nothing changes setting it on or off). Thanks

RichardNobel, 2021-03-17:
Same here! I had kinda taken it for granted, this loss of quality because of scaling. 🤔 

But if the text display can be improved, I'll have even more fun re-playing "[The Legend of Zelda: The Minish Cap](https://zelda.gamepedia.com/The_Legend_of_Zelda:_The_Minish_Cap)". 😃

xXS4ALLx, 2021-03-22:
Head over to the Discord channel (#powerkiddy-v90) and look for user "Consty", he recompiled gpSP emulator that has the screen filtering fix.

gameblabla, 2021-03-22:
> Head over to the Discord channel (#powerkiddy-v90) and look for user "Consty", he recompiled gpSP emulator that has the screen filtering fix.

it's not really a "fix" but just the addition of another filter, which looks better but is more CPU intensive too.
That may do it for some people but yeah, just saying :P

xXS4ALLx, 2021-03-22:
> it's not really a "fix" but just the addition of another filter, which looks better but is more CPU intensive too.
> That may do it for some people but yeah, just saying :P

Thanks for the clarification :)

xhnnsx, 2021-05-07:
hey, is there any fix for this problem? any chance to get the recompiled gpSP?
cheers and thank you

colibris79, 2021-05-18:
This would be a great improvement indeed. @xXS4ALLx @gameblabla I haven't been able to find that Discord channel or the recompiled version of gpSP, do you know where to get it?

nickalleye, 2021-07-27:
@colibris79 you can get to this server on the bottom of this page:
https://www.rghandhelds.com/

Anyway, does anyone know if there is a way to adjust the scaling manually?

j0eii, 2021-09-24:
pump this issue