TriForceX/MiyooCFW #144: Huge improvement over PocketSNES on Q90, V60, PocketGo

alex4one, 2021-03-18:
Hi, guys check this out: [https://boards.dingoonity.org/other-game-systems/pocketsnes-q90-v60-pocketgo/new/#new](url)
on MiYoo Pocket Go v1, stable 60 fps almost on all games... its amazing
Maybe you colaborate with author and help him with savestates/ sram, and push it in CFW v1.3.4-5)
Thanks

salvacam, 2021-03-18:
If the source code was accessible, anyone could try to help. But they haven't shared the code

clach04, 2021-03-21:
Thanks @alex4one for posting, I responded to that thread asking for source code. Hopefully they will respond.

I have an old (portable fork) at https://github.com/clach04/hot-pocketsnes but I'm unclear how this corresponds to the version on MiyooCFW.

TriForceX, 2021-03-21:
> 
> 
> Thanks @alex4one for posting, I responded to that thread asking for source code. Hopefully they will respond.
> 
> I have an old (portable fork) at https://github.com/clach04/hot-pocketsnes but I'm unclear how this corresponds to the version on MiyooCFW.

The one on the current CFW was just a compiled one from a random user, keep us informed if the one shared by alex has a repository available, it would be great.

shanedabes, 2021-04-15:
This sounds like it would be incredible for all of us, is it really dead in the water?

TriForceX, 2021-05-18:
> This sounds like it would be incredible for all of us, is it really dead in the water?

sadly the author have not replied yet... thats one of the problems when someone work "privately" projects that are supposed to be "open-source". 

We hope his answer soon since the emulator has some flaws (already mentioned in the main thread)

alex4one, 2021-05-18:
> sadly the author have not replied yet... thats one of the problems when someone work "privately" projects that are supposed to be "open-source".
>
> We hope his answer soon since the emulator has some [...]

Hi, i find his profile, last activity yesterday. [https://www.gp32spain.com/foros/member.php?2571-bitrider](url)
maybe it will help)

clach04, 2021-05-18:
@alex4one are you able to contact them to ask them for the source code? Thanks!

alex4one, 2021-05-19:
> @alex4one are you able to contact them to ask them for the source code? Thanks!

hi, no i dont know him/them. There is some problem, i dont know english well, and 0 in programming. I think it will be faster if somebody qualified contact him for more info. I just google him in forums.

TriForceX, 2021-05-19:
> hi, no i dont know him/them. There is some problem, i dont know english well, and 0 in programming. I think it will be faster if somebody qualified contact him for more info. I just google him in forums.

ive registered yesterday, that forum requires atleast 25 posts to start sending private messages, also his info is private there... if someone can still try in other ways it would be good, i tried and failed.

cricri-pingouin, 2021-05-25:
I've got an account on Dingoonity and can send PMs, so sent bitrider a PM asking him if he can join the discussion here or share the source code on Dingoonity.
Let's see!