TriForceX/MiyooCFW #145: resort/rename sections

clach04, 2021-03-21:
https://github.com/TriForceX/MiyooCFW/issues/138 has a few different suggestions. One is about renaming sections so emulators and games appear before "applications". Is this of interest for this project or is that considered user customization?

On twist to that original suggestion, instead or renaming to include a number, rename to "Tools".

BryMD, 2021-03-22:
Renaming 'Applications' to 'Tools' would be absolutely awesome! Would require GMenu2X to reroute 'Explorer' to the renamed 'Tools'-folder though...

MayanKoyote, 2023-12-29:
There is indeed logic and sence in this.
Now in "Applications" the only real self-sufficient applications are "Commander", "Simple Terminal", "FFPLAY" and "Explorer".
The rest of the shortcuts there are links to some sorts of tricks, such as a USB port mode switches (without explanation why the average user needs them), a TV output switch (the _most dangerous_ link in the world **), a Joypad mapper (?, it's not clear what it's doing except restarting GMenu2X), audio- and video- "recorders".
I.e. namely applied tools.
It would be appropriate to separate them into an own separate section. And "Tools" would be a good name for this.

 ** Seriously, if not to read the instructions (and no one really reads them), then to know that only holding down ‘SELECT’ will help you get out of an unpleasant situation with a black screen. A little more warnings about this would have been nice. As well as the mention of "'SELECT' (hold): Disconnect TV-out" when loading on one of the loading screens (of course, if the TV-out mode is enabled) since the selected mode is saved across reboots.

Apaczer, 2026-01-03:
The ActionLinks (Explorer, About etc.) are generated at runtime and bound to specific sections, so renaming these would be to much trouble IMO - you could hide them however with current gmenu2x. I will leave sorting as an option if favor: https://github.com/MiyooCFW/gmenu2x/issues/8

closed 2026-01-03 by Apaczer