TriForceX/MiyooCFW #146: Quake II Woes

BryMD, 2021-03-22:
Hi guys!

Do we know for sure that the Quake II port supports the full version PAKs?! I have 2 commercial versions of Quake II (one with just pak0.pak and one with pak0, pak1 and pak2.pak), and both break the Quake II port at launch.

Both Mission Packs require the full version PAKs to be installed. 'Mission Pack 1: The Reckoning' boots, but only to the console, while 'Mission Pack 2: Ground Zero' breaks at launch like the full version.

Would be good to know before grinding on ;)

Sincerely,
BryMD

gameblabla, 2021-03-22:
Hello, when i ported Quake 2 to it i only ever tried the shareware version. (and certainly none of the mission packs either. In fact, getting this to work at all was a pain)

BryMD, 2021-03-22:
@gameblabla You're 'sauce'?! You are officially a legend, hehe :D

The screen goes black for a second and then back to GMenu2X. pak0.pak alone is a hefty 184MB. May the file size itself pose a problem? As for the Mission Packs, no worries! I'm on it ;)

For reference: the command line parameters for the mission packs are '+set game xatrix' and '+set game hipnotic', and require their respective PAKs in folders named 'xatrix' and 'hipnotic' in the base Quake II folder.

Sincerely,
BryMD

closed 2021-10-15 by TriForceX