TriForceX/MiyooCFW #147: powkiddy v90 korean not work

sadducean, 2021-03-23:
in powkiddy v90 korean font is unreadable...
All korean text replace with question marks.

is there any solutions?

TriForceX, 2021-03-23:
are you using the cfw right?
that should be a character encoding problem, it can be solved.

sadducean, 2021-03-23:
> are you using the cfw right?
> that should be a character encoding problem, it can be solved.

Yes! i installed [Powkiddy-V90-Q90_CFW_v1.3.3_04_10_20.img] just downloaded here

all korean rom filename is boxed question marks.... eventhough selected korean on settings.
also korean text broken on settings menu

luxi78, 2021-03-31:
Chinese font also has the same problem

TriForceX, 2021-05-29:
Can you guys try to change the font-face manually to default from skin settings?

[image]

akirataguchi115, 2023-07-28:
Let's close this one for now and re-open if the issue persists?

Apaczer, 2024-01-26:
fixed with: https://github.com/MiyooCFW/gmenu2x/commit/c5a6beb9e5df8baee4f61cb2da780f33f85f7ac5

I've plan to add expanded keyboard with CJK characters, but thinking now how to make it less cumbersome for other users.

closed 2024-01-26 by Apaczer