TriForceX/MiyooCFW #153: Powkiddy V90 hotkey issue

demarkd, 2021-05-18:
I believe I read somewhere that power+start should bring up the poweroff menu on the V90, but either way when I perform that combination in the main menu on CFW 1.3.3, it locks up the system entirely. Holding down power+select (which does usually work) then does not work to perform a clean shutdown, forcing me to perform a dirty shutdown.

salvacam, 2021-06-03:
> I believe I read somewhere that power+start should bring up the poweroff menu (edit: take a screenshot) on the V90, but either way when I perform that combination in the main menu on CFW 1.3.3, it locks up the system entirely. Holding down power+select (which does usually work) then does not work to perform a clean shutdown, forcing me to perform a dirty shutdown.

I turn off the V90 by pressing and holding Start or pressing Start + L

salvacam, 2023-03-11:
These are the hotkeys for the next version
https://github.com/TriForceX/MiyooCFW/wiki/Miyoo-CFW#console-hotkeys

Apaczer, 2023-03-11:
and for gmenu2x while frontend is active see: <https://github.com/MiyooCFW/gmenu2x#controls>

closed 2023-03-11 by Apaczer