TriForceX/MiyooCFW #157: Feature Request: faster FAT32check

nanderer, 2021-05-24:
Hi,

from time to time the PowKiddy freezes or i just forget to shut it down probably. Its much better then the stock rom that you do some file system checking but its verly slow, indeed its faster to switch it of again, pull out the card, put it into my laptop  (running debian) and run sth like 

````
sudo fsck.vfat /dev/mmcblk4 -a
````

wait just a very short time and plug it back in to the powkiddy which then dosnt find a fat dirty bit and boots up quite right.

what do you do when you detect a dirty bit, some old software, or fsck.vfat with -unnessesery options like -f or stuff to check for bad blocks, it takes , i guess more then a minute , each time to check its by its own, maybe you can speed things a bit up even tho vfat dosnt support jornaling. 

As a mainly a GNU/Linux user i could also live with another better filesystem, but i have no expirience if extX is better then using sync-mounts, and cause its dose, and normaly i just switch the device of in pause mode its not necesery to do a full sweep, only when somesthing freezes in the middle of sometimes...

if it wouldnt be a read only embeded probab;y squasy thingy i could look into the init.d or so but i am not to much familiar with an incomplete gnu/linux system (im only learnt to work with LSB-Comform stuff) 

thanks for looking into this and keep up the good work

cricri-pingouin, 2021-05-25:
Yes, it's a bit of a bummer.
Although at the risk of hijacking your ticket, I'd rather see a change that once you're into the menu, there is no read/write access to the card (unless you change settings I suppose), so that if you simply switch off from the power switch, it doesn't corrupt the card. Also that would sort out the issue that sometimes I forget to slide the switch, and the next time, bummer, dead battery!

nanderer, 2021-05-25:
thanks for your reply

maybe its a good idea to access the fat partition a different way at runtime, didnt thought about it, but have you tried out mtools (tools that are similar to dos accessing fat-fs'es, used it back in the old days when reading/writing to floppys cause its much faster and feels more direct then doing mounts and unmounts), maybe only outside the "payload"-software, so just ro after its quits, show menu only after all syncs are done and then when writing new configs or so just write it to ramdisk and copy it over using mcopy...

but, again sorry, what exact command are you using for checking the fs, it feels like ms scandisk (that took also for ages, cause it was doing things very inefficient), i would be really interested if its the hardware or the different working algorithm that makes it taking for ages. Maybe you should not hide the output of the programm in use, than at lest i could follow it and see that its doing sth (possible just remove the > /dev/null 2>/dev/null at  the end of a line somewhere)...

thanks in advance

cricri-pingouin, 2021-05-25:
I didn't fix the card on a PC, I just let the Powkiddy do its own checks and fixes. It takes a while, but thankfully, I've only messed up my card once!
If it is a recurring problem for you, then I'd say you'd need to look at a proactive solution rather than reactive, i.e. try to figure out how to stop the card being corrupted rather than finding the best way to fix it.

my2iu, 2021-08-06:
Hey that's a good idea. Maybe the saved games and other writable stuff could be moved to the ext4 partition, which is more reliable. Then the fat partition could be mounted as readonly, so it would never be corrupted due to software issues. If you really want the save data and configuration files be accessible from Windows, they make a separate, smaller fat partition for just the writable data. That partition would be faster to scan if there are errors. If they could get fstrim to work, that could be used to improve the physical reliability of the sdcard too.

my2iu, 2021-08-07:
I'm not great with Linux stuff, but poking around the /etc/fstab, I noticed that the root partition isn't mounted with noatime. That option might reduce corruption issues on the sdcard a little bit. It might also be good to upgrade the mount type from ext2 to ext4 if the kernel supports it.

closed 2021-10-15 by TriForceX