TriForceX/MiyooCFW #158: Feature Request: auto-grow partition 4 if there is space left

nanderer, 2021-05-24:
just as the raspberry pi dose just after you imaged it to the card, you could include some code that grows the main partition ( partition 4 at /mnt/) at the first run. For me it was no problem to do that myself but for newbies it would be a good thing if you could strike this setup step after imaging the disk, and growing a fat32 dosnt requiere much code or computing power...

thanks a lot and keep up the good work

cricri-pingouin, 2021-05-25:
I just rebuilt a card this morning and though of this. But then I decided against it because:
- It would be a lot of work that I'd rather see spent fixing bugs or improving performance.
- Seeing that you would have installed "MiniTool Partition Wizard" anyway, and that your card is already in your PC, doing it there and then is not really more effort from the user.
- I imagine that doing it on the device would take a lot longer, and would be more risky ("MiniTool Partition Wizard" is tried and tested, and your PC is less likely to run out of battery mid way!).

nanderer, 2021-05-25:
thanks a lot for all the replies:
> Seeing that you would have installed "MiniTool Partition Wizard" anyway,
how are you seeing this, i have but mostly to manage my NTFS partitions, i don't like putting sdcards or other external memory with linux fs'es plugged into windows machines, allays to many misleading, i cant get it do you want to reformat it, dialogs that can be quite dangers. Also  the miniTool can take ages for initialising and scanning all disks, much worse then gparted, but parted is still the fastest...

of cause i would also categorise this as not as important then speeding up emulation or other things

cricri-pingouin, 2021-05-25:
I'm assuming that you're installing the CFW on the original card, so you will have to delete all the partitions, and the guide suggests to use "MiniTool Partition Wizard" to do so.
So when the subsequent step in the guide says to use the same tool to extend the partition, you should have it?
If you started from a blank card, I suppose you won't have to delete the partitions, but the next time you want to install a MiyooCFW update, then you'll have multiple partitions as well.
I'm sure there are other tools that let you extend the partition, but as I was following the guide, I just did what was suggested, and it was rather straightforward.
Also note what I said in my previous reply: it took several minutes on my PC using a USB 3.0 reader to extend the partitions and then copy the files to the extended partition, so I'd imagine that the same process on the console itself would take way longer! You might even run out of battery, so I'd say that doing it on a PC is probably quicker and safer anyway.

TriForceX, 2021-05-26:
This is already working on the upcoming next version, so tools like MiniTool to expand  mainpartition will not required anymore.

I will leave this issue opened until we release the next cfw version.