TriForceX/MiyooCFW #160: Picture Slideshow Viewer on the Pocketgo?

BryMD, 2021-05-25:
Hi everyone!

Anyone up for taking a jab at porting a picture slideshow viewer to the PocketGo?!

An application for displaying pictures (ideally with folder slideshow- and fullscreen downscaling-functions) on the device would truly complete the MiyooCFW in my view, and would provide us with the final vessel for enjoying all major categories of media (games, audio, video, graphics) on the same screen!

Picture slideshow viewers available for earlier consoles running GMenu2X include PV2X, PicShow and DinImaVie, to mention a few. Is it possible to port any of those to the MiyooCFW perhaps?

Keep up the amazing work!

Sincerely,
BryMD

TriForceX, 2021-07-25:
Yes is possible, i will see if someone can port it

BryMD, 2021-07-27:
Hi @TriForceX !

That would be **totally awesome**!!! I myself am **sincerely** missing a proper way to flip through pictures on the device! And for the simple fact that anything with a screen produced in the past 20-or-so years has had this ability, the CFW just wouldn't be complete without it ;)

I am well aware that GMenu2X has basic picture viewer capabilities through Explorer, but is quite too cumbersome to be called a feature (and suspect it to be more of a side effect of screenshot/box art displaying capabilities).

On top of my head, PV2X has the source code available for download: [https://sourceforge.net/projects/pv2x/](https://sourceforge.net/projects/pv2x/)

Thanks again for keeping the Miyoo CFW alive and kicking!

Sincerely,
BryMD

BryMD, 2021-07-30:
PS: I will, of course, chip in and design a proper icon for it once ported ;)

Sincerely,
BryMD

szymor, 2021-08-08:
I ported PV2X for MiyooCFW. Take a look [here](https://github.com/szymor/pv2x/releases). If you have any problems with it, please create a ticket in the relevant repository.

BryMD, 2021-09-13:
Freakin' awesome, @szymor!! And thanks a bunch @TriForceX for being the middle-man on this one!

The MiyooCFW has finally grown up and has become the trick of all trades I'd been desiring! Only bug I've found is that PV2X both rotates and MIRRORS the images when 'Vertical' and 'Best Fit' orientations are chosen. Not a problem for me personally, but have posted the issue in the PV2X fork itself for reference.

Thanks again, guys! You're LEGENDS!!

Sincerely,
BryMD

PS: As for a custom icon, I've found the o2xiv icon already present in most themes (including FontiGrid) to be quite sufficient for me personally. However, if a custom icon still is appreciated, I'll of course keep my end of the bargain ;)

[image]

closed 2021-10-15 by TriForceX