TriForceX/MiyooCFW #161: startup always showing message: Unclean shutdown deteched . Checking fat32 ....

dlee21, 2021-05-31:
startup always showing message: Unclean shutdown deteched . Checking fat32 ....

TriForceX, 2021-05-31:
How are you shutting down the console?

szymor, 2021-07-25:
I suggest closing it as the submitter probably turns off the console without proper shutdown.

BryMD, 2021-07-27:
Second @szymor on that!

On topic: This is my only major (unfixable) gripe with the PocketGo! I've bought 3 PocketGo-s: One for myself, one for my 4-year old son, and one for my best mate who is _far_ from the sharpest tool in the shed. I am predicting carnage and 24/7 CFW reinstalls as soon as the 2 latter receive their respective consoles :P

Toon-nooT, 2022-01-13:
Sometimes i don't have time to cleanly shut down, nor can i wait for the filesystem check.
I edited the script to wait 2 seconds for a keypress in case a unclean shutdown was detected. This will then skip the check.
(not all keys work though)

`#Check if fat32 is flagged as "dirty", and if so unmount, repair, remount
if dmesg | grep "mmcblk0p4" > /dev/null;  then
echo -e "\e[31mUnclean shutdown detected.\e[0m"
echo -e "\e[32mPress key to skip FAT32 partition check...\e[0m"
read -n 1 -t 10
	if ! [ $? == 0 ]; then
		# no key was pressed
		echo -e "\e[32mChecking FAT32 partition...\e[0m"
		umount /dev/mmcblk0p4
		fsck.vfat -y /dev/mmcblk0p4 > /dev/null; 
		mount /dev/mmcblk0p4 /mnt -t vfat -o rw,sync,utf8
		echo -e "\e[32mCheck complete.\e[0m"  	
	fi
fi`

aksenav2, 2022-09-07:
> 
Mine didn't turn on after that.

aksenav2, 2022-09-07:
> Sometimes i don't have time to cleanly shut down, nor can i wait for the filesystem check. I edited the script to wait 2 seconds for a keypress in case a unclean shutdown was detected. This will [...]

Mine didn't turn on after that.

Toon-nooT, 2022-09-07:
Did you at least see the prompt:
"Unclean shutdown detected"
Because only that part would have been changed.
If it really doesn't turn on, i suspect a problem on file or partition level...

aksenav2, 2022-09-07:
> 

> Did you at least see the prompt: "Unclean shutdown detected" Because only that part would have been changed. If it really doesn't turn on, i suspect a problem on file or partition level...

It is after deleting or changing according to the instructions that it stops turning on, hanging on the splash screen. But if I insert the mine backup file again, it turns on normally.

aksenav2, 2022-09-07:
> 

> I used Ubuntu in a VM to mount the partition. Edited and saved the file with default editor. I remember that you may have to take care of the line endings when you copy/paste it from here. You need to have a look if it still looks like before after copying.
> 
> I don't have the file anymore, because i'm now happily using this method: #302

I use linux mint and standard editor. Thanks for the advice, I'll try again. I always turn it off through the shutdown menu, but it happens that from some games and applications there is no exit button, for example, in a music player.

Apaczer, 2023-03-14:
We added [FAT check script](https://github.com/MiyooCFW/sdcard/blob/master/main/apps/fsck/fsck.sh) and an [option](https://github.com/MiyooCFW/sdcard#main-partition) to enable/disable it on startup. From my experiance if there's user's data corruption from unsafe shutdown it is not critical damage and may be fixed later after OS start. Might be worth just to add prompt at startup informing about possible corruption on BOOT/MAIN partition to suggest fsck.vfat

closed 2023-03-14 by Apaczer