TriForceX/MiyooCFW #164: Unable to Navigate in new firmware update.

AtticusAubert, 2021-07-03:
Hi guys, can someone be of assistance please. I have a Bittboy V2 or 2.5 (unsure). I've updated to the latest firmware though the buttons are not functioning properly. They are assigned to different functions which makes it difficult for me to access emulators. I can use the d-pad as their function is either a A or B button.

szymor, 2021-07-25:
Please be sure to use the CFW for your revision of the console. If it still happens, then probably you have an unsupported revision of hardware and need to wait for a developer willing to fix the issue for your device.

TriForceX, 2021-07-25:
> I have a Bittboy V2 or 2.5 (unsure)
you have to be sure before install a cfw version... anyways it seems youre using a wrong one in your console, thats why you have problems on controls, each console has different keyboard configuration, if you dont know which version is your console you should post images here to figure out

AtticusAubert, 2021-07-30:
Thanks for the feedbacks. I realised that I was installing the wrong cfw version. I own the V3 bittboy but thought it was a V2.

closed 2021-07-30 by AtticusAubert