TriForceX/MiyooCFW #166: Start up MIYOO title screen stuck (Q90 & Q20mini)

FirstTimeN00b, 2021-07-30:
Issue as per the title. i have load the firmware without any issue. 

The only thing out of the ordinary i could imagine is that it tried to create to create a G: and H: drive on the sd card as well as the boot E: and the main I:. is this a common issue, is there a solution out there.

FirstTimeN00b, 2021-07-30:
[image]

Looking on Minitool it only has two partitions the boot and main, is that correct?

looking at the original sd card there is a linux swap partition and a rootfs(Ext4) partition

[image]

coliff, 2021-08-04:
If you only have two partitions like the first image then you haven't set it up correctly. You need to burn the IMG to the microSD card with Win32DiskImager. You never need to manually create any partitions.
Follow the instructions in the setup guide: https://github.com/TriForceX/MiyooCFW/wiki/Custom-Firmware-Setup :-)

geezer500, 2021-09-22:
I used the Pocket Go and PowKiddy Q90-V90 image as listed in the guide/link mentioned by coliff for my Q20 mini.
Everything worked 1st time. No issues whatsoever.

closed 2023-03-14 by Apaczer