TriForceX/MiyooCFW #242: XYC Q8 ?

wgarnett, 2021-11-07:
I think driver maybe is this site , plz make a preview 1.3.4 CFW to support first Q8 release .... thx .

https://drive.google.com/drive/folders/15F4P9uKCGn5lPhkMSBVke8pjMkTMdbTm
or
http://drive.google.com/drive/folders/1sHdBiyfWhgUOLh0SCPZ73H3el6t1NCuJ

TriForceX, 2021-11-07:
We already are doing the proper test to add support to the XYC Q8. is already on the to-do list ([see here](https://github.com/TriForceX/MiyooCFW/discussions/231
)) for the next release.

The preview we have are only for some testers, is not made for public yet (due is a test),

We have adapted the most of stuff from the current Miyoo CFW, but there are some issues that need to be solved.

- Sounds is working only through headphones.
- Screen has a granulated filter.
- Buttons needs to be remapped because is a mess at this moment (but is working).

Some test i did yesterday when i received the console: https://streamable.com/4m5enz

The other cfw images around there are based on the old cfw for the bittboy v2.5 (very old) so is useless in the latest miyoo cfw, thats why we are adapting the current cfw to be compatible with the Q8

Apaczer, 2023-03-11:
support added for next release, activated with "CONSOLE_VARIANT=xyc" in console.cfg

closed 2023-03-11 by Apaczer