TriForceX/MiyooCFW #243: Stucked on the start screen (Q90)

BlaupunktU47, 2021-11-08:
Hello folks, first of all, am a dumbass (am sorry)

I bought my powkiddy 2 days a go and I stay in my school for kinda 9 hours, and trying to keep more battery, I turn the screen brightness on the settings lower has possible, than, my problem started, even trying to reset, i cound't see anything, i tried remove the battery pack, even put a lantern on the screen to try see something, nothing worked.

then I've done the most stupid thing that I could do. basically I put in the computer school, searched in his filles and found the  "suniv-f1c500s-miyoo.dtb", and saw the line "backlight 2pwm", and I tried to edit turning up to 50, then, my screen got stucked.

then I tried anything to reinstall the CFW again but I always got stuck on win32 image diskreader with the error No. 5, and am scared of maybe f*cked up my console or the SD card. So... what I can do? pls help me

TriForceX, 2021-11-09:
Have you tried another SD card? the sd card that came with the console is very bad and got corrupted easily.
Also have you checked the image file you downloaded has the correct size? because some people have experienced some issues installing it on the SD card due the downloaded file was uncomplete (usually happens when you download large files with a bad connection).

PS: suniv-f1c500s-miyoo.dtb is a compiled file, you shouldnt edit it

Apaczer, 2023-03-14:
files at BOOT partition shouldn't be modified, except for config files or adding new modules.

closed 2023-03-14 by Apaczer