TriForceX/MiyooCFW #244: Instead of turning the screen off, display "It is now safe to turn off your device" or whatever on shutdown.

nfriedly, 2021-11-18:
I keep finding my V90 "on", but with a dead battery because I had previously done a software shutdown but forgot to flip the switch.

Instead of completely shutting down the system, could we instead flush and unmount the disk and then display something on the screen to remind people like me to actually flip the power switch?

TriForceX, 2021-11-19:
Something like the old windows?
[image]

nfriedly, 2021-11-19:
Yeah, exactly, haha!

akg83, 2023-05-24:
as long as the shut down sequence has to involve multiple steps, I think having a feature like this (if possible to properly implement) can be a great idea

TriForceX, 2024-05-27:
for the current behaviour (power-off from menus) can we delay the power-off command showing a picture + timer before continue power-off? (previously shutting down most of services?)

btw https://forums.raspberrypi.com/viewtopic.php?t=274936#p1666287