TriForceX/MiyooCFW #245: fix "Date & TIME" saving functionality + add "fake clock" at a startup

Apaczer, 2021-11-23:
Within older version of Miyoo (eg. _NxHope_) if you set date/time to specific value it stays on that number regardless of resetting device. However with latest _Miyoo_ it goes back to its stock value every time. 
Afther that there could be added simple prompt before boot asking to set correct time/date with every startup (optional). Wich could act as a "fake clock" function.

- [ ] fix saving "clock&date" value after reboot
- [ ] add prompt at a startup to set current time/date

Cerezk, 2021-12-01:
Not sure that a "fake"/simulated clock is what we need, but this is the same reason I came here today to leave a bug report -- so I'm seconding this issue. The internal/hardware clock in the CFW doesn't seem to be working either on the PocketGo or on the PowKiddy Q90. It saves the changes, but then defaults back to factory default time every time the device is switched off and turned on again (and by that I mean it goes back to the same date and time, somewhat bizarrely in 2019 for me).

I suspect it may be a Linux OS/firmware settings issue?

Apaczer, 2021-12-01:
True - there is no internal clock. That's why the simulated clock may be useful to someone, but for others it might seem unnecessery step (for that I suggest to add it as an option eg. in a separate script). 

Benefit would be easy game progress tracking (savestates time stamp) and some games use clock to set proper in-game time.

Cerezk, 2021-12-01:
As an optional feature, I think it's a brilliant suggestion -- something that can be turned on or off in the settings.

I know it would be very useful with PokeMini's Breeder game for example -- which relies on the internal clock for its track of time for the Tamagotchi-like progression in the game.

But how to keep track of time while the device is switched off? It would need to have software keep on running in the background, for lack of hardware to do the job.

Alternatively what you're suggesting, the prompt at the startup to set the time again would be an option, but how is that different to setting the time in the game? Wouldn't it get tedious, and prove not practical, having to set it every time?

Apaczer, 2022-08-17:
First issue resolved and I'll be adding simple hotkey dialog for changing quickly date at anytime in gmenunx (with next patch release).

closed 2022-08-17 by Apaczer