TriForceX/MiyooCFW #247: Problem with flashing on Mac OS

atriot, 2021-11-26:
HI thanks for the guide. It is really useful. It's really difficult to do this on a mac! I've formatted with the SD Card Formater and tried to flash the CFW file into the SD card. But when I try to transfer the ROMS I discovered that there was a weird unknown space usage in the SD Card which I couldnt find. I have attached a screenshot for refeerence here https://ibb.co/CvqBSRy. Any Help would be greatly appreciated. Thanks everyone!

andrigamerita, 2021-11-26:
The main partition doesn't only hold your roms, but also apps, emulators, and native games (in the folders with those names), which all take up around 1GB.
You can just delete the folder of an emulator or a game you don't need.

Also, I don't know how big your SD card is, but have you expanded the main partition to the full space of the card, with a partitioning tool? If not, you have some amount of wasted free space on your SD card.

atriot, 2021-11-26:
Hi @andrigamerita thanks so much for replying. My SD Card is 16GB. There are 2 partitions. 
Here is a screenshot of the disk info and the contents of the folder: https://imgur.com/a/cdVAwNp
BTW Is there a good partitioning tool you can recommend? I am currently just using a free app called SD Card Formatter. thanks!

andrigamerita, 2021-11-26:
> There are 2 partitions. Here is a screenshot of the disk info and the contents of the folder: https://imgur.com/a/cdVAwNp

There should also be a "rootfs" partition on the card, if you flashed the IMG without doing anything strange, probably macOS just doesn't see it because it is ext4. But the boot and main partitions look normal, as they are in the IMG. So the card should work as it is and the console should boot, I don't know if you already tried.
In main/games there are various native games that take up space. If you don't play them, you can delete them. Same for main/emus, if you don't emulate a particular console you can delete the folder for a specific emulator.

> My SD Card is 16GB.

So yeah, you have roughly 12GB of unused space at the end of the card, you should expand the main partition with a partitioning tool, by using the unused space.

> BTW Is there a good partitioning tool you can recommend?

For macOS I know none, and I don't remember if the tool included with macOS lets you expand FAT32 partitions, it should. If not, I suggest [GParted](https://gparted.org). It's only for Linux but the website offers ISO files you can burn on a CD or flash on a pendrive, reboot your computer to it, and doing the partitioning job.

atriot, 2021-11-27:
> There should also be a "rootfs" partition on the card, if you flashed the IMG without doing anything strange, probably macOS just doesn't see it because it is ext4. But the boot and main partitions [...]

Hi thanks again for that. I've tried using the SD card on the v90 and it works. the problem is just that there is no rom in it as I do not have space to transfer the roms. I have run disk utility on the SD card heres how it looks: https://imgur.com/a/B2Vt6q3

I am not sure how to fix the partitions properly as i am using a basic app called SD Card Formatter which works with the mac to format the new SD card.

So I guess the problem now is about the partitioning on the SD card which I dont know how to fix....

atriot, 2021-11-27:
> Hi @atriot :)
> 
> I myself now EXCLUSIVELY use Mac OS for all things Miyoo, much thanks to Hans Luijten's absolutely BRILLIANT and FREE utility 'ApplePi-Baker' (https://www.tweaking4all.com/software/macosx-software/applepi-baker-v2/).
> 
> The utility is made with the Raspberry Pi in mind, but works FLAWLESSLY with BittBoy/PocketGo/PowKiddy as well - and does EVERYTHING needed from the same interface: Format, Restore, Resize (<- the solution to your current problem) and Backup.
> 
> Contemplated a while to make a step-by-step guide, but it's literarily so easy and straight forwards that none is needed :P
> 
> [image]
> 
> Sincerely, BryMD

Hi Thanks for the info. I've tried downloading the app but it looks like this one mine. https://imgur.com/tE7Kh0O

I tried mounting with this app but the same problem with space (as above) occurred. What did you use to format your SD card? Thanks again

andrigamerita, 2021-11-27:
> I have run disk utility on the SD card heres how it looks: https://imgur.com/a/B2Vt6q3
@atriot it looks like Disk Utility just.. Doesn't acknowledge the unused space? I know nothing about the software BryMD suggested, you are better off flashing the GParted ISO onto a thumbdrive and rebooting your PC into that just to expand the SD partition

And also now that I'm thinking about it, I don't know if the CFW includes classic CLI tools for partitioning, if so you could use the included app "SDL Terminal" and do everything with command line from the console itself.. Yeah, even if it was possible, it would not be a pleasant experience at all.. But this gives me an idea: the CFW could include a script you can just start from the menu to expand the partition automatically
I will probably create an issue about this, it would be a nice improvement to the CFW

BryMD, 2021-11-27:
Hi again @atriot !

I use ApplePi-Baker for EVERYTHING!

1. Format SD Card: ApplePi-Baker
2. Install Custom Firmware: ApplePi-Baker
3. Expand Main Partition: ApplePi-Baker
4. Backup SD Card: ApplePi-Baker

I see you've downloaded an old version (1.9.9). Version 1.9.9 is the last version WITHOUT partition expansion support. The download link above the one you clicked is version 2.2.4, and is the one you want!

@andrigamerita : Partition auto-expansion has been implemented I believe, and hopefully is on track for release with CFW 1.4.0...

Sincerely,
BryMD

BryMD, 2021-11-28:
Hi again, @atriot :)

In danger of presuming wrongly, I think that the main problem here is separating what is a disk from what is a partition/volume. As such, I'll try to explain it briefly:

Imagine the whole DISK of your Micro SD card as being a big box. Then imagine putting a much smaller box inside that box. The much smaller box is the main PARTITION/VOLUME of the custom firmware where all your stuff needs to go. The operating system ONLY sees this small box available for storage. As such, what we need to do here is to expand the small box to make it fill the entire big box.

This is where ApplePi-Baker (version 2.0 and up ;) ) and step 3 in my post above comes in :)

Hope this helps, and my apologies of presuming wrongly...

Sincerely,
BryMD

danielnixon, 2021-11-29:
> Edit: @andrigamerita : I read somewhere that partition auto-expansion was being considered and/or been implemented, and hopefully is on track for release with CFW 1.4.0...

https://github.com/TriForceX/MiyooCFW/discussions/223#discussioncomment-1480622

atriot, 2021-11-30:
> > I have run disk utility on the SD card heres how it looks: https://imgur.com/a/B2Vt6q3
> 
> @atriot it looks like Disk Utility just.. Doesn't acknowledge the unused space? I know nothing about the software BryMD suggested, so I can't help with thay, you are better off flashing the GParted ISO onto a thumbdrive and rebooting your PC into that just to expand the SD partition
> 
> And also now that I'm thinking about it, I don't know if the CFW includes classic CLI tools for partitioning, if so you could use the included app "SDL Terminal" and do everything with command line from the console itself.. Yeah, even if it was possible, it would not be a pleasant experience at all.. But this gives me an idea: the CFW could include a script you can just start from the menu to expand the partition automatically I will probably create an issue about this, it would be a nice improvement to the CFW

thanks fr the 

> Hi again @atriot !
> 
> I use ApplePi-Baker for EVERYTHING!
> 
>     1. Format SD Card: ApplePi-Baker
> 
>     2. Install Custom Firmware: ApplePi-Baker
> 
>     3. Expand Main Partition: ApplePi-Baker
> 
>     4. Backup SD Card: ApplePi-Baker
> 
> 
> I see you've downloaded an old version (1.9.9). Version 1.9.9 is the last version WITHOUT partition expansion support. The download link above the one you clicked is version 2.2.4, and is the one you want!
> 
> Sincerely, BryMD
> 
> _Edit: @andrigamerita : I read somewhere that partition auto-expansion was being considered and/or been implemented, and hopefully is on track for release with CFW 1.4.0..._

Hi everyone. thanks so much again for the reply. Thanks for the patience as I am not very tech proficient.
Just an update, I couldnt download the newer version of applepibaker because my OS is still on catalina and i cant clear up enough space on my mac to update to latest OS. So I guess I'm just giving up on this for now.... either that or I will try and find a PC which can do this. thanks again!

BryMD, 2021-11-30:
Hi again @atriot :)

I have to say you've gotten me profoundly confused with your last post. ApplePi-Baker runs PERFECTLY on Mac OS 10.15 Catalina as well...

To put it bluntly, I PERSONALLY had bigger difficulties following the instructions for Windows than doing it blindly with ApplePi-Baker - hence my excitement about this BRILLIANT piece of software!

Perhaps a step-by-step guide is warranted after all... :P

Sincerely,
BryMD

For everyone else on a Mac, here's the download page again: [https://www.tweaking4all.com/software/macosx-software/applepi-baker-v2/](https://www.tweaking4all.com/software/macosx-software/applepi-baker-v2/)

Cerezk, 2021-12-01:
A past Mac user here (migrated/upgraded to Linux). As far as I know, the included Disk Utility application with Mac OS X and its variants does support basic formatting of partitions/volumes to FAT32, but it has the same bugs/quirks as the default disk utility in GNOME -- it just isn't able to perform more advanced operations such as resizing on Windows partitions.

So what you'd need to find, @atriot, for the resizing/expanding is a custom disk utility for Mac based on Gparted and its code. Something that is a little more serious/advanced than the included Disk Utility on the system.

Rezzy-dev, 2021-12-31:
@TriForceX, @35o125

Guide for Mac users:

**Steps to flash an SD card:**

1. Check the "/dev/disk" number of your SD card in Disk Utility.
2. Open the Terminal app
3. Unmount the card using this command in the Terminal: `diskutil unmountDisk "/dev/DISK"` (Where DISK is the disk# of your SD card.)
4. To write the disk image to the card: `sudo dd if="IMAGE.img" of="/dev/DISK"` (Where IMAGE is the full file path to the Miyoo CFW disk image you've downloaded, and DISK is the assigned disk# address of your SD card in Disk Utility.)

_**Warning:**_ Make sure that you enter the correct disk number for your SD card, or you will be overwriting a different card/disk on your computer!! Double-check the number in Disk Utility, or run the `diskutil list` command in the Terminal, and look for your card and its assigned number by the size of the disk.

Note: When you hit Return/Enter on the above command, dd/Terminal will appear to do nothing. This is normal. Wait until dd finishes its work, and it will give you a summary of the successful write. This may take quite some time, so be patient. If you hold CTRL and press T on your keyboard, it will show you how dd is working.

**Steps to back up your whole SD card into a disk image file:**

1. Make sure you have enough space on your hard drive -- as much as the whole SD card is in size
2. Unmount the card (see above)
3. Open the Terminal app
4. Reverse the from/input ("if") and to/output ("of") values in the previous write command: `dd if="/dev/DISK" of="IMAGE.img"` (See what IMAGE and DISK stand for above.)

**Steps to expand the size of the "main" partition on the card:**

The easiest way to do this, if you have a CD/DVD drive, is to burn the GParted LiveCD ISO to an optical disc, and boot from it on your Mac by turning on the Mac and holding down the C key on your keyboard at the boot sound/chime.

When successful, you'll be greeted with an old-type, Windows-like blue/system screen. Just leave all the settings as default and hit the Return/Enter until it loads up GParted.

(Note: You'll likely need a wired mouse and keyboard for this, as your wireless/Bluetooth mice and keyboards will not work here! Wireless controllers don't work outside of the operating system.)

[image]

1. By default, GParted will load your Mac's hard drive first. Make sure you switch to your SD card using the top-right dropdown menu button!

2. Right-click the partition labelled "main" on your SD card, and choose the "Resize/Move" option.

3. In the resize window that pops up, drag the slider to the right, expanding the selection to the entire space on the card, and hit the Resize button.

4. Now double-check that everything looks right (with only the "main" partition on the SD card expanded in size), and if everything looks correct, apply your changes by clicking the green tick button in the menu bar, or Edit > "Apply All Operations". You may need to confirm your changes -- check that they are correct and what you really want to do. At this point GParted will actually operate on your changed disk volumes and make the changes you've instructed it to make.

5. When done, close GParted and reboot your Mac. You've successfully resized your SD card, and your SD card is ready for use. Safely eject it from your Mac, and pop it into your Miyoo device.

TriForceX, 2021-12-31:
> @TriForceX, @35o125
> 
> Guide for Mac users:
> 
> **Steps to flash an SD card:**
> 
>     1. Check the "/dev/disk" number of your SD card in Disk Utility.
> 
>     2. Open the Terminal app
> 
>     3. Unmount the card using this command in the Terminal: `diskutil unmountDisk "/dev/DISK"` (Where DISK is the disk# of your SD card.)
> 
>     4. To write the disk image to the card: `sudo dd if="IMAGE.img" of="/dev/DISK"` (Where IMAGE is the full file path to the Miyoo CFW disk image you've downloaded, and DISK is the assigned disk# address of your SD card in Disk Utility.) You will need to enter your administrative password to confirm, and hit the Return/Enter key on your keyboard.
> 
> 
> _**Warning:**_ Make sure that you enter the correct disk number for your SD card, or you will be overwriting a different card/disk on your computer!! Double-check the number in Disk Utility, or run the `diskutil list` command in the Terminal, and look for your card and its assigned number by the size of the disk.
> 
> Note: When you hit Return/Enter on the above write command, dd/Terminal will appear to do nothing. This is normal. Wait until dd finishes its work, and it will give you a summary of the successful write. This may take quite some time, so be patient. If you hold down CTRL and press T on your keyboard, it will show you how dd is working.
> 
> **Steps to back up your whole SD card into a disk image file:**
> 
>     1. Make sure you have enough space on your hard drive -- as much as the whole SD card is in size
> 
>     2. Open the Terminal app
> 
>     3. Unmount the card (see above)
> 
>     4. Reverse the from/input ("if") and to/output ("of") values in the previous write command: `dd if="/dev/DISK" of="IMAGE.img"` (See what IMAGE and DISK stand for above.)
> 
>     5. Wait until dd finishes and gives you the (successful) write summary
> 
> 
> **Steps to expand the size of the "main" partition on the card:**
> 
> The easiest way to do this, if you have a CD/DVD drive, is to burn the GParted LiveCD ISO to an optical disc, and boot from it on your Mac by turning on the Mac and holding down the C key on your keyboard at the boot sound/chime.
> 
> When successful, you'll be greeted with an old-type, Windows-like blue/system screen. Just leave all the settings as default and hit the Return/Enter until it loads up GParted.
> 
> (Note: You'll likely need a wired mouse and keyboard for this, as your wireless/Bluetooth mice and keyboards will not work here! Paired, Bluetooth/wireless controllers don't work outside of the operating system.)
> 
> > You can download the GParted LiveCD ISO here: https://gparted.org/download.php
> > Choose the "amd64" (64-bit) version, as all Intel Macs are 64-bit. (Only old PowerPC Macs are 32-bit.)
> 
> [image]
> 
>     1. By default, GParted will load your Mac's hard drive first. Make sure you switch to your SD card using the top-right dropdown menu button!
> 
>     2. Right-click the partition labelled "main" on your SD card, and choose the "Resize/Move" option.
> 
>     3. In the resize window that pops up, drag the slider to the right, expanding the selection to the entire remaining space on the card, and hit the Resize (OK/confirm) button.
> 
>     4. Now double-check that everything looks right (with only the "main" partition on the SD card expanded in size), and if everything looks correct, apply your changes by clicking the green tick button in the menu bar, or Edit > "Apply All Operations". You may need to confirm your changes -- check that they are correct and what you really want to do. At this point GParted will actually operate on your changed disk volumes and make the changes you've instructed it to make.
> 
>     5. When done, close GParted and reboot your Mac. You've successfully resized your SD card, and your SD card is ready for use. Safely eject it from your Mac, and pop it into your Miyoo device.

Wiki updated! thanks!

Rezzy-dev, 2022-01-01:
Thanks @TriForceX. We can probably close this issue, now, since a working, tested and supported solution has been found, and a Mac setup/flashing guide, using the built-in Mac OS software, has been added to the official instructions.

closed 2022-01-05 by TriForceX