TriForceX/MiyooCFW #248: main partition is FAT-formatted

agentroadkill, 2021-11-28:
I used dd (VDD from homebrew) to write the PowKiddy Q90 English firmware to my SD car. The main partition on the card is FAT (not FAT32 or exFAT). The partition is limited to 1.5 GiB and can’t be resized: is this expected? Copying main off the card and changing the partition to a larger FAT32 or exFAT causes the console to not boot.

TriForceX, 2021-11-29:
The firmware version included in the package SD cards are modified versions of our firmware, usually modified by the manufacturers, so the format, partitions, and the files there usually are not the same as the firmware stuff we provide. We cannot guarantee any functionallity with the stuff we dont provide (like the stuff that came with the console)

The SD must be FAT32, not FAT or exFAT. Writting the CFW image on the sd will create 4 partitions, the MAIN partition is the right one to expand after the image is written.

agentroadkill, 2021-11-29:
Sorry I should have been more clear. I used a new 128 GiB SD card to flash, following the instructions and download link provided [here](https://github.com/TriForceX/MiyooCFW/wiki/Custom-Firmware-Setup#pocket-go-and-powkiddy-q90-v90) for my PowKiddy V90. Agree FAT seems wrong - I was curious if using (g)dd rather than the Windows tool in the instructions could make a difference, or if there was a problem with the image I happened to download?

Cerezk, 2021-12-01:
Is it FAT for sure? Because that's very old (we're talking archaic, DOS era). Is it not FAT32, but for some reason it's being reported as just FAT? Where are you getting the indication that it is FAT? In Disk Utility?

35o125, 2021-12-08:
I have exactly the same issue. 
Disc Utility doesn't let me expand the main partition, so I can't load ROMs. I've tried inserting flashed SD card with the new firmware into a V90 and it works, just empty without ROMs. How to increase the space on the partition? Please!

35o125, 2021-12-08:
Thank you.
Installation process is a bit confusing for a noob like me. There are several links for download. Any clue which one to get?

Rezzy-dev, 2021-12-08:
If you're on a modern Mac you'll be wanting the amd64 one, as that means 64-bit:
https://downloads.sourceforge.net/gparted/gparted-live-1.3.1-1-amd64.iso

i686 is 32-bit (which no new Mac can run, only old Macs with PowerPC processors).

Rezzy-dev, 2021-12-08:
If your Mac has a CD drive, simply burn the ISO image with either Disk Utility or Toast to a CD:
https://www.macinstruct.com/tutorials/how-to-burn-an-iso-disc-image-on-your-mac/

35o125, 2021-12-08:
I did that. Now I have a new CD. Have no idea how to run it.

Rezzy-dev, 2021-12-08:
Then restart your computer, holding down the C key on your keyboard. Sometimes you need to shut down the computer and start it up from cold, as the boot process is a little buggy on Mac OS X. After you hear the chime, hold down C until the computer starts booting from the CD.

Rezzy-dev, 2021-12-08:
When successful, you will see the computer boot into a light version of Linux that starts the GParted disk formatting and partitioning utility.

Make sure you back up your computer before you start using GParted to resize/reformat the SD card, and, again, make sure you are NOT making changes to the computer's internal hard drive, but only the inserted SD card.

Rezzy-dev, 2021-12-08:
Find the FAT32 volume on the SD card labelled "main", and expand it using GParted to the full size of the card. Then right click and "apply" the changes. At this point GParted will write the changes to the SD card and operate on the partition. (It will ask for a confirmation from you that this is really what you want to do before it does it.)

35o125, 2021-12-08:
Mac got frozen at 'don't touch keymap' stage (configuring console data).

Rezzy-dev, 2021-12-08:
> Mac got frozen at 'don't touch keymap' stage (configuring console data).

On booting from the CD?

Rezzy-dev, 2021-12-08:
You need to select "don't touch keymap", and then hit the "OK" button:
[image]

35o125, 2021-12-08:
it booted from the CD, showed some tiny text with commands. I've hit esc, and then enter for default. After a long script a windoe appeared with 'Don't touch keymap' line marked in red. I couldn't do anything at that point other than force shutdown. Couldn't hit the OK button.

Rezzy-dev, 2021-12-08:
Did you try using the arrow keys on the keyboard or the Tab key?

Rezzy-dev, 2021-12-08:
You just need to leave "down't touch keymap" selected and move to the "OK" button and hit the Enter key on your keyboard.

Rezzy-dev, 2021-12-08:
Then go for the default language, mode and other settings and press Enter.

35o125, 2021-12-08:
Yeah, one more reason to hate Mac. Thanks for staying with me! Will update.

Rezzy-dev, 2021-12-08:
This is what GParted looks like:
[image]

Rezzy-dev, 2021-12-08:
> Yeah, one more reason to hate Mac. Thanks for staying with me! Will update.

No worries. Let me know if you run into any issues. Better to ask questions than mess things up. ;)

35o125, 2021-12-08:
Will do :)

Rezzy-dev, 2021-12-08:
Once you find the "main"-labelled, FAT32 partition on the SD card, simply select and right click it, and choose "Resize/Move" in the context menu. Then you'll get the resize visual interface above, where you can just drag the resize slider to the right.

And then you just need to apply the operations/changes (writing them to the SD card).

Rezzy-dev, 2021-12-08:
You switch to the SD card with the top right selection button. Be advised that default, the first disk that GParted loads is your internal hard drive. So you most **definitely** want to switch to the SD card before you change anything.

The SD card will have 4 partitions, labelled "boot", "rootfs", "swap", and "main".

If you've made changes to the wrong disk, don't panic, simply quit and shut down GParted and the computer **without** applying the changes.

Rezzy-dev, 2021-12-09:
Wireless/Bluetooth keyboard users, note that your Bluetooth keyboard may not work here. You may need to plug in a USB, wired keyboard.

35o125, 2021-12-09:
Thank you. I actually had to go and buy a wired mouse.
Anyways, managed to get in, but can't see my SD card in the top right corner. It only shows my HD.

Rezzy-dev, 2021-12-09:
> Anyways, managed to get in, but can't see my SD card in the top right corner. It only shows my HD.

That's odd. Hmm... how is the SD card plugged into the computer? Try plugging the SD card in only **after** you load GParted. See if that makes a difference in terms of it showing up. I suspect that Mac OS may have hijacked (or turned off) the SD card on boot.

35o125, 2021-12-09:
plugged at the back. it can be seen in finder.
tried inserting several times while in GParted, no change. Tried refreshing device, nothing..

35o125, 2021-12-09:
> This is what GParted looks like: [image]

how do you have 931Gb drive selected top right, and yet you can see those 4 partitions on the left side, plus resizing?

Rezzy-dev, 2021-12-09:
> plugged at the back. it can be seen in finder.
> tried inserting several times while in GParted, no change. Tried refreshing device, nothing..

Hmm... can you share a screenshot with me of what Disk Utility shows you about the SD card's partitioning in Mac OS?

When you say you plugged it in in the back of the computer, do you mean you're using a USB adapter for the SD card?

Rezzy-dev, 2021-12-09:
> how do you have 931Gb drive selected top right, and yet you can see those 4 partitions on the left side, plus resizing?

This is just an example screenshot I shared with you just to show you what GParted looks like, and it's showing the computer's hard drive, not the SD card. It's also not my computer.

35o125, 2021-12-09:
No, I am just using adapter for micro card. I have a slot at the back of my mac.

35o125, 2021-12-09:
[image]

Rezzy-dev, 2021-12-09:
This is how GParted actually looks like with the SD card loaded in it on my Linux computer:
[image]

35o125, 2021-12-09:
> This is how GParted actually looks like with the SD card loaded in it on my Linux computer

I envy :)

Rezzy-dev, 2021-12-09:
> [image]

Hmm... looks to be in order. Mac OS can't read the rootfs partition, because that's a Linux formatted (EXT4) partition, and the swap partition (that's why it's displaying those funny "disk" names), but your SD card seems to be fine.

35o125, 2021-12-09:
new firmware works in the console, it's just ROMs free : l

Rezzy-dev, 2021-12-09:
I wonder why the SD card is not showing up on the GParted LiveCD. I'm guessing it would be a driver issue -- a missing driver for the internal SD card reader for the Mac. Although that's very rare to happen on Linux, and something that would only really happen when running Linux on a Mac.

Do you happen to have a USB reader/adapter for the SD card lying about?

35o125, 2021-12-09:
nope.

Rezzy-dev, 2021-12-09:
Hmm... :/

Rezzy-dev, 2021-12-09:
What about giving that other utility a go, on Mac OS?

Because we'll have a tough time getting a driver for the Mac SD card reader running on the LiveCD.

Rezzy-dev, 2021-12-09:
Alternatively, you could always install Linux on a USB flash drive, and boot from it on your Mac. That will allow you to install the necessary drivers, and you'll have a version of Linux you can just take with you.

35o125, 2021-12-09:
I've tried reading through it and I don't understand much of it

Rezzy-dev, 2021-12-09:
> I've tried reading through it and I don't understand much of it

On how to install Linux to a USB thumb drive? Or the Mac utility?

35o125, 2021-12-09:
> > I've tried reading through it and I don't understand much of it
> 
> On how to install Linux to a USB thumb drive? Or the Mac utility?

ApplePi-Baker

Rezzy-dev, 2021-12-09:
> ApplePi-Baker

Ah. Yeah, I've never tried it, but it seems simple enough to use from the screenshots.

35o125, 2021-12-09:
I've never worked with Linux

Rezzy-dev, 2021-12-09:
Do you want to try creating a USB bootable version of Linux for your Mac with GParted on it?

It's farily easy to do using UNetbootin: https://en.wikipedia.org/wiki/UNetbootin

I can walk you through it.

35o125, 2021-12-09:
I have install ApplePi but dunno how to expand the partition
[image]

Rezzy-dev, 2021-12-09:
Umm... Right click "main", and choose "resize" or "expand" from the context menu? Maybe...

35o125, 2021-12-09:
[image]

Rezzy-dev, 2021-12-09:
But we can see here that "main" is properly formatted as a FAT32 partition -- so that's a great sign! It means that gdd did do its job properly after all -- and answers the original question in this thread. It's not FAT, it's FAT32.

35o125, 2021-12-09:
I am not the OP

Rezzy-dev, 2021-12-09:
> [image]

Hmm... :/

What about in the Mac menu for the app? On top.

Rezzy-dev, 2021-12-09:
> I am not the OP

Well, you did ask for support in this issue/thread. :P

35o125, 2021-12-09:
[image]

Rezzy-dev, 2021-12-09:
Wow... looks much more confusing to use than in the screenshots... o.O

35o125, 2021-12-09:
I think I've just erased the sd

35o125, 2021-12-09:
fck

Rezzy-dev, 2021-12-09:
> I think I've just erased the sd

Ouch... How? :o

[1230 more replies]