TriForceX/MiyooCFW #251: Wrong key mapping in FCEUX

usta83, 2021-11-30:
I have installed the V90 CFW from Miyoo on my Q20 mini. In the menu the keys are fine, but when I open FCEUX A & B are switched. So confirming works with B instead A. In the Game its ok, because I have mapped it right. But e.g. saving with R+A doesnt work. I have to do R+B.

Does anyone have a solution for this?

andrigamerita, 2021-11-30:
@usta83 From what I can see, you can open the file `.fceux/fceux.cfg` (it's on the `main` partition of the SD card) with a text editor, scroll down to the lines saying `SDL.Input.GamePad` and swap the numeric values for 0A and 0B. So put whatever number is assigned to 0A, to 0B, and viceversa.

usta83, 2021-11-30:
I tried it and something has changed within the game. But not a simple switch between A and B. Also in the Fceux menu B was still the confirm button (instead of A)

Cerezk, 2021-12-01:
I can confirm this. The A-B switched state seems to be hardcoded into FCEUX, and it seems to have something to do with the BittBoy's button configuration -- for which this build of the software was originally developed for. B is A on the BittBoy, and Y is B -- hence the confusion.

But FCEUX's internal/menu button configuration should really follow the user's settings for the in-game button configuration in terms of the A (accept/confirm) and B (cancel) buttons.

Cerezk, 2021-12-01:
FCEUX is also not the only emulator on the system to have this issue. Ohboy and PokeMini have it, too. (And I think PCSX ReARMed as well, if I remember correctly...)

Cerezk, 2021-12-01:
[image]
[image]
[Note the button layout/labelling change as these devices evolved.]

usta83, 2021-12-01:
So this means that I should change the values like that:

A->B
TA ->A

Or is there no solution?

I could live with the in-game adjustment but I always get confused with speed saving and loading states - R+A & R+B
And this is not my only device. So I always get confused when I switch devices.

Cerezk, 2021-12-01:
I believe there is no solution at the moment (without editing the source yourself and recompiling), because this button behaviour is hardcoded and the FCEUX code needs to be rewritten to support multiple button layouts -- because this same software runs both on BittBoy and on later devices such as the PocketGo and PowKiddy Q90 and V90 devices.

I think the everyone has been working on this, but it's a slow and gradual progress to adjust and to rewrite all the code to support the new button layout as well. Some of the emulators support it, now, and have the correct A-B behaviour in the menu, some don't.

Also it's:
A->B
B->Y

The position and wiring of the buttons is the same, but the labelling has changed -- hence why this is so confusing, both for the developers and for the users. All software ported to and developed for the original BittBoy that took the BittBoy's button layout for granted and that we're using on these later devices now will now have this A-B button switch in their menu/core until updated/fixed.

Cerezk, 2021-12-01:
And yes, I know how you feel -- I have the same problem, I get confused with the interface when playing different games in different emulators.

usta83, 2021-12-01:
Thank you for your detailed response!

In a nutshell: changing the .cfg won't help me right?

Cerezk, 2021-12-01:
I think the only reason T2 (TA) works as the cancel button is because the emulators have been written to support both B/Y and T2 (TA) as the cancel button.

Cerezk, 2021-12-01:
> In a nutshell: changing the .cfg won't help me right?

Correct. Because the button behaviour in the menu is hardcoded into the software with FCEUX. I tried changing the config file as well, and it only changes the in-game button behavior/mapping, not the menu behaviour.

usta83, 2021-12-01:
Thanks a lot!

I hope there will be some development soon. I could not find the stock firmware for my Q20. Otherwise I would try that.

Cerezk, 2021-12-01:
No worries. :) I can't imagine that the stock firmware would have it any different -- all this comes from the same development and source code. The stock firmware will only have an earlier version of the code, I imagine (at least it did with my PocketGo and PowKiddy Q90). Unless they went in and made the changes without contributing back to the source.

Cerezk, 2021-12-01:
"Development" depends on all of us. This is open source, a community effort. It's work that needs doing, and people who want to do it. :)

Cerezk, 2021-12-01:
@usta83, there is an updated version of FCEUX currently in beta testing, called FCEUX Improved. I haven't checked it out yet, but maybe someone has tried to fix the button layout issue in the menu for it?

https://github.com/TriForceX/MiyooCFW#beta-testing

You could give it a whirl and see (if you're interested, you feel up to it and have the time).

usta83, 2021-12-01:
I have tested this improved beta and I had the feeling the performance was worse. So I had deleted it. 

But I'm gonna try it again concerning the mapping

Cerezk, 2021-12-01:
There's also another revision of FCEUX there in beta testing, called FCEUX Rumble, by Steward.

usta83, 2021-12-01:
I will try this!

Cerezk, 2021-12-01:
If neither of these have the fix, I may look into it at some point myself, as I'd like to fix this on my device and for all of us. Just need to find the time.

usta83, 2021-12-01:
@Cerezk No chance.... I tried all of them. All have the same problem.

It would be so awesome if u can solve the problem!

usta83, 2021-12-01:
One more question:

Why is the Fceux menu on my RG300 different? E.g. I am missing the L1&L2 Buttons for fast Forward on my Q20. I do use this very often on my RetroFW device.

Cerezk, 2021-12-02:
> @Cerezk No chance.... I tried all of them. All have the same problem. I have taken a picture of my current setting. Your explanation makes absolutely sense now. [image]
> 
> It would be so awesome if u can solve the problem!

Thanks for testing them, @usta83. I'll have a look into it as soon as I can. (I'm currently studying up on C.) Once I figure out the code and how to compile it, I'm sure I can figure out and implement a solution that works for all devices.

Cerezk, 2021-12-02:
> One more question:
> 
> Why is the Fceux menu on my RG300 different? E.g. I am missing the L1&L2 Buttons for fast Forward on my Q20. I do use this very often on my RetroFW device.

I'm not sure, but since RetroFW is a different firmware distribution, perhaps they've changed something in their FCEUX build? You'd better as there, ask them why this feature is missing from their build:

https://github.com/retrofw/retrofw.github.io/issues

Alternatively, if RetroFW is also based on the same version of Linux and an ARM processor/architecture, you could try copying the Q20 FCEUX over to your RG300 and see if it works, and if the buttons are still missing. Worst case scenario is that it will simply not start on that system.

usta83, 2021-12-02:
thanks a lot!! You would do all here a big favor:-)

I mean the fast forward works on the RG300 and not in my Q20. I don't understand why in the Q20 CFW an apparently old version is used.

Cerezk, 2021-12-02:
> I mean the fast forward works on the RG300 and not in my Q20. I don't understand why in the Q20 CFW an apparently old version is used.

Hmm... Is the RG300 one an OPK, or a standard Linux binary (folder)? If it's the latter, you could try the reverse, and copy the RG300 one over to the Miyoo/Q20 to see if it works. It's worth a try.

I wonder why those changes weren't sent back here, though. It's a bit dodgy that in order to move forward with new devices, we seem to keep abandoning/neglecting older firmware releases by separating things into different, independent projects. There's no reason why the emulators can't progress at the same pace for all platforms. But that's just my two cents on this.

usta83, 2021-12-02:
Yes it's an OPK... I will give it a try, but I am not very confident in my skills.

Cerezk, 2021-12-02:
> Yes it's an OPK... I will give it a try, but I am not very confident in my skills.

Ah, never mind then, it probably won't work. I've tried extracting an OPK or two and running them on the Miyoo platform, but none of my efforts have been successful so far. I think the dependencies for the software between the two systems are very different.

usta83, 2021-12-02:
If you say so ;-)

I hope u find some time for an update

Cerezk, 2021-12-02:
I'm actually confused as to where the source is located, because there's no link to it from the front page.

I've found something, but I'm not sure if this is the source code the current version of FCEUX on Miyoo CFW 1.3.3 was compiled from. @TriForceX?

usta83, 2021-12-02:
> I'm actually confused as to where the source is located, because there's no link to it from the front page.
> 
> I've found something, but I'm not sure if this is the source code the current version of FCEUX on Miyoo CFW 1.3.3 was compiled from.
> 
> https://github.com/forbbs/miyoo_fceux
> 
> @TriForceX?

Have u found anything?

Rezzy-dev, 2021-12-02:
> Have u found anything?

Not yet. First I want to make sure it's the right source I'm looking at...

Rezzy-dev, 2021-12-03:
Found it! At the end of fceu.h:
https://github.com/forbbs/miyoo_fceux/blob/master/src/fceu.h

`#define JOY_A   1
#define JOY_B   2
#define JOY_SELECT      4
#define JOY_START       8
#define JOY_UP        0x10
#define JOY_DOWN        0x20
#define JOY_LEFT        0x40
#define JOY_RIGHT       0x80`

These are the default controls set, used in the menu.

Rezzy-dev, 2021-12-03:
I would need to somehow check for whether the device it is running on is not a BittBoy, and then swap the values of JOY_A and JOY_B on that condition.

usta83, 2021-12-03:
> Found it! At the end of fceu.h: https://github.com/forbbs/miyoo_fceux/blob/master/src/fceu.h
>
> #define JOY_A 1 #define JOY_B 2 #define JOY_SELECT 4 #define JOY_START 8 #define JOY_UP 0x10 #define [...]

Great!!!

I hope you can make it! Is not TA & TB missing there?

Rezzy-dev, 2021-12-03:
> Is not TA & TB missing there?

These are the default controls from the get-go, for the menu -- so it won't have the full set. I believe the value for JOY_B (2) is actually the T2/TA (later A) button here. By swapping JOY_A and JOY_B's values, we get the correct behaviour for button A and B on non-BittBoy devices. But I need to give it a go and test it, see.

Rezzy-dev, 2021-12-03:
Ah, I found another source for FCEUX for OpenDingux:
https://github.com/retrofw/fceux

It may be that the Miyoo CFW FCEUX was built from the RetroFW source.

In which case I may be wrong about OPKs not working on Miyoo CFW. If you can unsquashfs them (unpack them), the folders and the binary they contain should be able to run just the same on Miyoo CFW in theory. So it's possible that the RetroFW copy of FCEUX you have (the RG300 one) will be able to run on the Miyoo CFW (the PowKiddy Q20). You just need to unpack the OPK and place the files in the correct location on the Q20.

Rezzy-dev, 2021-12-03:
OH! Actually, that's branched from here:
https://github.com/dmitrysmagin/fceu320-rzx50

Have a look at the button configuration here -- it's more up-to-date and features the shoulder buttons!

Rezzy-dev, 2021-12-03:
It's possible that if we compile this original and most up-to-date source, the GCW Zero version, we'll get exactly what we're looking for, actually!

Have a look at the GCW Zero's button layout:
[image]

It has exactly the same buttons and overall layout as these post-BittBoy devices. And it also runs OpenDingux.

Rezzy-dev, 2021-12-03:
Hmm... I've tried downloading and running a few OpenDingux software on the Miyoo CFW, but I get execution errors with all of them -- none of them actually run on Miyoo CFW. So I'm quite confused as to what's going on. Is the Miyoo CFW not based on OpenDingux? Does it vary from OpenDingux so much that it can't run OpenDingux programs/applications? There's virtually no documentation on the differences between these operating systems and builds, so it's very difficult to know what's going on unless you've been there, unless you have insider knowledge/insight.

usta83, 2021-12-03:
Oh that doesn't sound good... I hope you'll get the answer soon

usta83, 2021-12-03:
> OH! Actually, that's branched from here: https://github.com/dmitrysmagin/fceu320-rzx50
> 
> Have a look at the button configuration here -- it's more up-to-date and features the shoulder buttons!
> 
> This may be the reason the Miyoo version is missing them -- the current build is way behind the actual source code.

Do you think an update is possible? I guess only the Beta version were possible

Rezzy-dev, 2021-12-03:
I'll attempt to build both sources I have here, and we'll see what happens. If either of them runs, there can be hope for fixing the A-B button issue in FCEUX. I'm not sure about the shoulder buttons, but if I can see what they've done in the later release, perhaps I can port it back to this one.

usta83, 2021-12-03:
I'm wondering why nobody else here is supporting. I would have guessed many people are annoyed of this issue...

Rezzy-dev, 2021-12-03:
I'm guessing they would have heard it a lot by now, yes...

The thing is it's not a big issue from a developer's perspective -- because trust me, there are much bigger issues/bugs to deal with -- but it is something that would impact a lot of users in terms of the smoothness of the user experience, especially while still getting used to the platform.

So it would be a common complaint, and requiring quite a bit of work and time to fix for each emulator. Work that's not very challenging or rewarding, just tedious.

And since these emulators come from all these different sources/platforms, they'll all have their own quirks, according to the platform they come from. So it's just the nature of the system to have these small inconsistencies in terms of the user experience.

That said, it doesn't mean it can't be done, provided there are people willing to put time and effort into this cleaning work.

It's a bit like how working office spaces never get cleaned unless there is someone whose dedicated job is to clean it up.

Rezzy-dev, 2021-12-04:
Okay, I've attempted to build them from source using the provided toolchain, @usta83, and both builds failed with a whole bunch of undefined references:

/bin/sh: 1: sdl-config: not found
src/file.o: In function `FCEU_fopen(char const*, char const*, char*, char*, int, char const**)':
file.cpp:(.text+0x2fc4): undefined reference to `gzopen'
file.cpp:(.text+0x3024): undefined reference to `gzgetc'
file.cpp:(.text+0x30ac): undefined reference to `gzseek'
file.cpp:(.text+0x30ec): undefined reference to `gzread'
file.cpp:(.text+0x30f4): undefined reference to `gzclose'
src/state.o: In function `FCEUSS_SaveMS(EMUFILE*, int)':
state.cpp:(.text+0x1364): undefined reference to `compress2'
src/state.o: In function `FCEUSS_LoadFP(EMUFILE*, ENUM_SSLOADPARAMS)':
state.cpp:(.text+0x1ad0): undefined reference to `uncompress'
src/video.o: In function `SaveSnapshot()':
video.cpp:(.text+0x834): undefined reference to `compress'
src/video.o: In function `SaveSnapshot(char*)':
video.cpp:(.text+0xae8): undefined reference to `compress'
src/netplay.o: In function `FCEUNET_SendFile(unsigned char, char*)':
netplay.cpp:(.text+0x1f4): undefined reference to `compress2'
src/netplay.o: In function `NetplayUpdate(unsigned char*)':
netplay.cpp:(.text+0x3bc): undefined reference to `uncompress'
src/drivers/dingux-sdl/input.o: In function `GetKeyboard()':
input.cpp:(.text+0x360): undefined reference to `SDL_GetKeyState'
src/drivers/dingux-sdl/input.o: In function `GetMouseData(unsigned int (&) [3])':
input.cpp:(.text+0x398): undefined reference to `SDL_GetMouseState'
src/drivers/dingux-sdl/input.o: In function `FCEUD_UpdateInput()':
input.cpp:(.text+0x3fc): undefined reference to `SDL_PollEvent'
input.cpp:(.text+0x438): undefined reference to `SDL_PollEvent'
input.cpp:(.text+0x444): undefined reference to `SDL_GetKeyState'
input.cpp:(.text+0x660): undefined reference to `SDL_GetKeyState'
input.cpp:(.text+0x674): undefined reference to `SDL_GetKeyState'
input.cpp:(.text+0x688): undefined reference to `SDL_GetKeyState'
input.cpp:(.text+0xa64): undefined reference to `SDL_GetKeyState'
src/drivers/dingux-sdl/input.o:input.cpp:(.text+0xb6c): more undefined references to `SDL_GetKeyState' follow
src/drivers/dingux-sdl/dingoo.o: In function `FCEUD_GetTime()':
dingoo.cpp:(.text+0x11c4): undefined reference to `SDL_GetTicks'
src/drivers/dingux-sdl/dingoo.o: In function `main':
dingoo.cpp:(.text.startup+0x24): undefined reference to `SDL_Init'
dingoo.cpp:(.text.startup+0x29c): undefined reference to `SDL_Quit'
dingoo.cpp:(.text.startup+0x3f4): undefined reference to `SDL_Quit'
dingoo.cpp:(.text.startup+0x920): undefined reference to `SDL_Quit'
dingoo.cpp:(.text.startup+0xbe0): undefined reference to `SDL_Quit'
dingoo.cpp:(.text.startup+0xd5c): undefined reference to `SDL_GetError'
dingoo.cpp:(.text.startup+0xd88): undefined reference to `SDL_Quit'
dingoo.cpp:(.text.startup+0xd90): undefined reference to `SDL_Quit'
src/drivers/dingux-sdl/dingoo-sound.o: In function `WriteSound(int*, int)':
dingoo-sound.cpp:(.text+0xc4): undefined reference to `SDL_LockAudio'
dingoo-sound.cpp:(.text+0x12c): undefined reference to `SDL_UnlockAudio'
src/drivers/dingux-sdl/dingoo-sound.o: In function `SilenceSound(int)':
dingoo-sound.cpp:(.text+0x168): undefined reference to `SDL_PauseAudio'
src/drivers/dingux-sdl/dingoo-sound.o: In function `KillSound()':
dingoo-sound.cpp:(.text+0x17c): undefined reference to `SDL_CloseAudio'
dingoo-sound.cpp:(.text+0x184): undefined reference to `SDL_QuitSubSystem'
src/drivers/dingux-sdl/dingoo-sound.o: In function `InitSound()':
dingoo-sound.cpp:(.text+0x23c): undefined reference to `SDL_InitSubSystem'
dingoo-sound.cpp:(.text+0x250): undefined reference to `SDL_AudioDriverName'
dingoo-sound.cpp:(.text+0x72c): undefined reference to `SDL_OpenAudio'
dingoo-sound.cpp:(.text+0x73c): undefined reference to `SDL_PauseAudio'
dingoo-sound.cpp:(.text+0x790): undefined reference to `SDL_GetError'
src/drivers/dingux-sdl/dingoo-video.o: In function `KillVideo()':
dingoo-video.cpp:(.text+0x30): undefined reference to `SDL_FreeSurface'
src/drivers/dingux-sdl/dingoo-video.o: In function `BlitScreen(unsigned char*)':
dingoo-video.cpp:(.text+0x420): undefined reference to `SDL_LockSurface'
dingoo-video.cpp:(.text+0x470): undefined reference to `SDL_UnlockSurface'
dingoo-video.cpp:(.text+0x47c): undefined reference to `SDL_Flip'
src/drivers/dingux-sdl/dingoo-video.o: In function `dingoo_clear_video()':
dingoo-video.cpp:(.text+0x6e8): undefined reference to `SDL_MapRGBA'
dingoo-video.cpp:(.text+0x6f8): undefined reference to `SDL_FillRect'
dingoo-video.cpp:(.text+0x700): undefined reference to `SDL_Flip'
dingoo-video.cpp:(.text+0x71c): undefined reference to `SDL_MapRGBA'
[... 50 lines cut ...]
font.cpp:(.text+0x38): undefined reference to `SDL_MapRGB'
font.cpp:(.text+0x48): undefined reference to `SDL_SetColorKey'
src/drivers/dingux-sdl/gui/font.o: In function `KillFont()':
font.cpp:(.text+0x70): undefined reference to `SDL_FreeSurface'
src/drivers/dingux-sdl/gui/font.o: In function `DrawChar(SDL_Surface*, unsigned char, int, int)':
font.cpp:(.text+0xcc): undefined reference to `SDL_UpperBlit'
src/utils/crc32.o: In function `CalcCRC32(unsigned int, unsigned char*, unsigned int)':
crc32.cpp:(.text+0x0): undefined reference to `crc32'
src/utils/crc32.o: In function `FCEUI_CRC32(unsigned int, unsigned char*, unsigned int)':
crc32.cpp:(.text+0x4): undefined reference to `crc32'
src/utils/unzip.o: In function `unzReadCurrentFile':
[... cut ...]
Makefile:244: recipe for target 'fceux' failed
make: *** [fceux] Error 1

Since there's no detailed documentation on the build process, I have no idea how to proceed from here -- I'm at a dead end. I don't know whether these sources or even the toolchain are up-to-date and in sync! I don't know anything about this platform... :/ :(

So yeah, your best bet is still to try and unsquashfs (unpack) the RetroFW version of FCEUX, and try it out on Miyoo to see if it runs. Otherwise, unless we get some help, I'm out of ideas. :/

usta83, 2021-12-04:
Well... U tried it... 

Thank so much for your efforts!!!

Rezzy-dev, 2021-12-04:
Okay, good news and bad news:

The good news is that after spending a long two hours to compile the toolchain from the latest Miyoo source, I _was_ able to successfully compile one of the sources for FCEUX that I had found here! It has successfully cross-compiled, and an ARM binary for the Miyoo platform has been created.

The bad news is that the generated binary, while it actually ran on the Miyoo system -- which is always a great sign -- it didn't manage to intitialise the video drivers and crashed. I'm thinking that this has something to do with this source being too old and having outdated video drivers.

I will try the other, more up-to-date source tomorrow, or the closest to it when I have some time to spare...

It would be really great to get some accurate pointers as to which source code these binaries were originally compiled from for the current release of the Miyoo CFW. That would make my job so much easier! I wouldn't have to dig and hunt to find the right combination of sources, and risk having something that is out of date or behind the current release. There are much better ways to spend my effort and time -- like actually fixing up FCEUX!

But we're getting closer now.

And for anyone else reading this, coming into this and trying to compile some software for the Miyoo -- do **not** use the provided compiled package for the toolchain, as it is way too outdated, and strong chances are will **not** work (like it didn't for me). I don't think it should even be linked to from the Wiki if its build is so outdated -- as that only helps to further complicate/confuse things for anyone new coming into the scene.

Rezzy-dev, 2021-12-04:
Hold on, no! Something funny is going on here...

It _didn't_ crash because of the video driver! It actually worked as it's supposed to! But GmenuNx for some reason refused to launch the new binary. I could run it from the Terminal, but not from GmenuNx -- even though the shortcut I created for it was exactly the same as the working one for the shipped version of FCEUX here! @_@

Really odd. Not sure what's going on...

usta83, 2021-12-04:
Sounds promising!! I'm curious:-)

Rezzy-dev, 2021-12-05:
*Sigh.* Both sources compiled successfully with the new, up-to-date toolchain, and both will launch/run in the Terminal on Miyoo but not in the main interface, in GmenuNX. :/

I really don't know what's going on. There's no indication whatsoever that there is any error, and yet the compiled binaries with the toolchain won't work on the Miyoo system.

At this point I think it may be easier to hexedit the existing binaries to make the changes we need than trying to recompile them -- which is quite sad, but yeah...

usta83, 2021-12-11:
Have tried to hexedit the binaries yet?

Rezzy-dev, 2021-12-11:
> Have tried to hexedit the binaries yet?

Yup. No luck. :/ I couldn't find the hexadecimal representation for the A and B button inputs.

usta83, 2021-12-11:
;-( 

I was so excited that someone is finally taking care. Thank you!

Rezzy-dev, 2021-12-11:
And I was excited to find a way to fix this. But I received no help and guidance (so far) either. Not even directions to the right source code. :/ :(

usta83, 2021-12-11:
Exactly... That's so sad...

Rezzy-dev, 2021-12-11:
I should probably take the time to point out that something is not open source if the source code for it is not easy to find and available. And that without basic documentation on how the video output works on the Miyoo platform, and what needs to be done to make a compiled emulator run on it, hardly anyone coming in will be able to do anything in terms of development on the platform. :/

Rezzy-dev, 2021-12-12:
Messing around with DOSBox, I managed to find out something:

The A and B buttons on Miyoo actually have the same keycodes as the Alt and Fn key on the computer keyboard, respectively.

http://en.qi-hardware.com/wiki/Key_codes

A -> Alt - 308
B -> Function-Key - 306

So I need to be looking for Alt, not A in the software's code. I need the hexadecimal values for the Alt keyboard input/key.

Rezzy-dev, 2021-12-12:
Correction: the B button (306) is on some keyboards the Fn key, and on some the left Ctrl key.

So they're Alt (A) and Ctrl (B).

Rezzy-dev, 2021-12-12:
Here is how the buttons correspond to keyboard keys on my PowKiddy Q90:

L: Tab
R: Backspace

X: Left Shift
Y: Space
A: Left Alt
B: Left Ctrl

Start: Enter
Select: Esc

Reset (Minus): Right Ctrl

D-pad: Arrow Keys

(And I have a strong suspicion that the extra Plus button on my PowKiddy corresponds to the Right Shift key.)

This also helps to shed light on why some applications behave unusual/strange on Miyoo (such as exiting on pressing the Select button) -- because they were originally created with a computer keyboard in mind.

usta83, 2021-12-23:
I found out that TriFoce is active on this discord channel: https://discord.gznetwork.com/

Maybe u can ask him/her there...

[37 more replies]

closed 2022-09-09 by Rezzy-dev