TriForceX/MiyooCFW #253: PICO-8 Image

gladel, 2021-12-07:
In Gameshow theme, PICO-8 seemed out of place from the rest of the images.

[image]

Rezzy-dev, 2021-12-08:
Ah, it's displaying the Pico-8 icon from the Default theme over the Gameshow graphics/splash. This happens when there's no icon image for Pico-8 in Gameshow's "icons" folder. It defaults to using the icons from the Default theme (understandably).

Go into gmenu2x/skins/GameShow/icons/ on the SD card, and make a duplicate of one of the (empty graphic) icons there, and rename the duplicate to pico8.png.

This will make the default Pico-8 icon disappear on the Gameshow skin splash for Pico-8.

Apaczer, 2021-12-08:
indeed @Rezzy-dev! Thanks for that explanation, I was going nuts over that icon overlaping all my external app's backdrops. Was confused to that point that I've deleted Default icons only to make sure the Gameshow skin work properly.

Rezzy-dev, 2021-12-09:
> indeed @Rezzy-dev! Thanks for that explanation, I was going nuts over that icon overlaping all my external app's backdrops. Was confused to that point that I've deleted Default icons only to make sure the Gameshow skin work properly.

Note that you don't need to delete it, you can just rename the Default theme's folder.

The only catch with this (renaming or deleting Default) is that using any of the skins, if an icon is missing from the skin, it will show a blank label for that app/program in GmenuNX. In the Gameshow skin, if you're missing and icon _and_ a background graphic for it, it will show the default background, and you won't know what app/program it is.

This is why this backup (revert to Default skin icons) feature exists, to ensure that apps/programs on GmenuNX will always have at least one identifiable graphic/icon for you to know what app/program you've selected.

Rezzy-dev, 2021-12-09:
Here are the Default icons -- for reference and if you wish to restore them:
[ICONS.zip](https://github.com/TriForceX/MiyooCFW/files/7681796/ICONS.zip)

Rezzy-dev, 2021-12-09:
Each icon's name corresponds to the executable binary's name for that program/application in /mnt/emus/, /mnt/games/, or /mnt/apps/.

E.g. The icon for /mnt/emus/pico8/pico8.elf will be "pico8.png".

closed 2021-12-09 by gladel