TriForceX/MiyooCFW #281: Working for Q8 v1.6

wgarnett, 2022-01-08:
Boot Section and LCD Driver Working 
https://www.mediafire.com/file/yoncf04wbbyqn7d/Q8.7z/file

TriForceX, 2022-01-10:
Yea we have that also working (q8 v1.2 and v1.6), the only thing thats need to be finished is the key mapping...
Btw this also need to be compatible (and made) with the current/latest cfw kernel source!

wgarnett, 2022-01-11:
you mean gmenu2x input.conf ?
I use this on my Q8 

up=keyboard,273
down=keyboard,274
left=keyboard,276
right=keyboard,275
modifier=keyboard,32
confirm=keyboard,308
cancel=keyboard,306
manual=keyboard,304
section_prev=keyboard,9
section_next=keyboard,8
pageup=keyboard,9
pagedown=keyboard,8
settings=keyboard,13
menu=keyboard,27
backlight=keyboard,51
power=keyboard,305
inc=keyboard,32
dec=keyboard,304

TriForceX, 2022-01-11:
No, this https://github.com/MiyooCFW/kernel/blob/f281eb848273b32f8ea5a2eb72f26791721fad9e/drivers/input/keyboard/miyoo_kbd.c#L41-L115

wgarnett, 2022-01-11:
someone get right lcd driver and merge into https://github.com/steward-fu/q8  ..... 
and then I can play unofficial version well ....

TriForceX, 2022-01-11:
Yea we have another builds working around there on the Q8 (including the one form Steward) but thats not the main task, our current source code should handle the Q8 along the other Miyoo consoles, thats the main purpose. i didnt had the time to adapt the current source code with the Q8 stuff, thats why the Q8 isnt still added on the official list. 

No one at this moment have helped with this, there are few people who got the Q8 stuff working but using an old source code, not the new one, and thats where we need to have it working.

salvacam, 2022-01-11:
> http://c.tieba.baidu.com/p/7691824426 , I play now .... using top Q8.7z file

What image did you use to replace the files in the Q8.7z file?

Do all the buttons work for you in the games and emulators and the sound?

wgarnett, 2022-01-11:
steward-fu/q8  ... replace 7z files with it .....
all work well on my Q8

salvacam, 2022-01-11:
> steward-fu/q8 ... replace 7z files with it ..... all work well on my Q8

I tried that yesterday, and it didn't work for me, the image looked bad
I'm going to try again

wgarnett, 2022-01-11:
7z files only match for q8_od_jckl_no_roms  .....
write image to MicroSD  and then replace all files  in the partition FC3000 ....
on China , there are 4 people now work fine (all 1.6 firmware) ...... 
but I know Q8 has 3 LCD version ......

wgarnett, 2022-01-12:
not finish loading =>  change another SD card or reboot more times .... until success 
 2 people and I was the same before ....

salvacam, 2022-01-12:
I have rewritten the image q8_od_jckl_no_roms.
Now it starts me up, but I have the same result when replacing the files in the FC3000 partition with the Q8.7z file. It looks bad

You can check that the content of the Q8.7z file you uploaded is correct, the mininit and rootfs files are the same size as in the image.

If the content you have uploaded is correct, I have another screen on my console (version 1.6 firmware)

wgarnett, 2022-01-12:
@salvacam  I recheck Q8.7z , it's right ..... 
if mininit and rootfs the same size , but maybe some byte hex is different  ..... just replace all on FC3000 partition ....

and I have a new question , anyone know how to change boot screen ??? I want to change a picture that I like myself

salvacam, 2022-01-12:
I think the image needs to be recompiled
Although here they explain a way
[https://www.reddit.com/r/Bittboy/comments/d6yxm8/tutorial_setting_custom_boot_splash_screen_on_the/](https://www.reddit.com/r/Bittboy/comments/d6yxm8/tutorial_setting_custom_boot_splash_screen_on_the/)

wgarnett, 2022-01-13:
@salvacam , thanks , and now do your Q8 work ???

salvacam, 2022-01-13:
No. I have gone back to do the same process a couple of times and I have the same result.
It looks like the picture I showed you.

I am doing something wrong or my screen is different from yours

I save the image q8_od_jckl_no_roms and copy the files from Q8.7z to the FC3000 partition

Can you record an image from your SD card and share it for testing?

wgarnett, 2022-01-13:
@salvacam , full image => https://www.mediafire.com/file/qg27qq7pllykwok/q8_od_fix.7z/file

salvacam, 2022-01-13:
> @salvacam , full image => https://www.mediafire.com/file/qg27qq7pllykwok/q8_od_fix.7z/file

Thank you very much, with that image it works perfectly

[image]

How do I control the volume and brightness of the screen?

In gmenu With L + Select I can modify the volume but later in games it is heard at maximum.
It would be nice to be able to modify the audio and the brightness in gmenu and when it is playing

wgarnett, 2022-01-13:
sorry , I don't know , some emulator setting can modify the audio  ....

wgarnett, 2022-01-13:
I use headphone , I'll use control on headphone to adjust audio  volumn

wgarnett, 2022-01-13:
@salvacam , I follow reddit guide , but fail to change boot splash image .... do you have other way and work ?

salvacam, 2022-01-13:
> @salvacam , I follow reddit guide , but fail to change boot splash image .... do you have other way and work ?

No, I don't know of any other way to change the splash image. Sorry

Another user with version 1.2 of XYC Q8 has tried the image with the fix and it works fine

salvacam, 2022-01-13:
I have combined your image with the English image of Steward
[https://www.mediafire.com/file/dhle7s05fhtgzai/q8_od_jutleys_no_roms_fix.zip/file](https://www.mediafire.com/file/dhle7s05fhtgzai/q8_od_jutleys_no_roms_fix.zip/file)

wgarnett, 2022-01-15:
@salvacam , great , thx image ... 
I try some method still not found where to replace boot splash picture ....

salvacam, 2022-01-15:
@wgarnett Thanks to you. Now you can use more emulators on my Q8
I don't know how to change the splash image. I don't know how the firmware works

wgarnett, 2022-01-16:
@salvacam  , thanks .... I test all work fine , except dosbox load wrong game and j2me half screen 
do you have better way to run ?

salvacam, 2022-01-17:
> @salvacam , thanks .... I test all work fine , except dosbox game only half screen do you have better way to run ?

it looks good to me in the v90 and the q8
[image]

I am using dosbox from cfw 1.3.3

@wgarnett  Check emulator/dosbox/dosbox.conf file
I have this
```
[sdl]
fullscreen=true
fulldouble=false
fullresolution=320 x 240
windowresolution=320 x 240
output=surface_dingux
autolock=true
sensitivity=100
waitonerror=true
priority=higher,normal
mapperfile=mapper.txt
usescancodes=false

[render]
scaler=none

[autoexec]
mount c /mnt/roms/DOS
C:
CD \LAUNCHBX
LAUNCH.BAT
```
Check that you have the same resolution defined

wgarnett, 2022-01-17:
@salvacam , setting is same .... I think maybe some game problem ...... I'll try some others , thanks.

wgarnett, 2022-01-18:
https://dos-bin.zczc.cz/%E9%87%91%E5%BA%B8%E7%BE%A4%E4%BE%A0%E4%BC%A0.zip , this game I try only 1/4 game screen show , can you work fine ?????

Apaczer, 2022-01-28:
@wgarnett 
>... I try some method still not found where to replace boot splash picture ....

What splash image is being displayed at startup? Do you have *.png of it or decoded Hex string?

wgarnett, 2022-01-28:
@Apaczer  I can't find Hex string of boot partition , so I can't change splash image

Apaczer, 2022-01-28:
If the image is 16bit format (rgb565) and the screen resolution is 320x240, then the binary size should match. From there you should look for any similarites in decompiled Hex and look for similar string in size (I assume it would be easier if you had that image file).

wgarnett, 2022-01-28:
@Apaczer  , splash image is here , https://github.com/steward-fu/q8/releases
I think path is "drivers/video/logo" located in the kernel directory

Apaczer, 2022-01-28:
I can see there only 80x80px Framebuffer boot-up logos, is that what you mean?

wgarnett, 2022-01-29:
@Apaczer

Splash image is like this if you decompile it from logos

https://miro.medium.com/max/1400/0*swBdePfPQlN1Ufm-

Apaczer, 2022-02-17:
> I have combined your image with the English image of Steward https://www.mediafire.com/file/dhle7s05fhtgzai/q8_od_jutleys_no_roms_fix.zip/file

Use this image and set your OFFSET in Hex editor to 4C064 (use guideline from reddit), the splash image string probably resides there but I can't be 100% sure (between 4C064 and 71864 blocks). Please be aware that you can damage your SD card this way.

wgarnett, 2022-02-18:
> 4C064

@Apaczer
4C064  is right , it work , thanks

sun6560032, 2022-03-29:
> 我已将您的形象与Steward https://www.mediafire.com/file/dhle7s05fhtgzai/q8_od_jutleys_no_roms_fix.zip/file 的英文形象相结合

Do you have the screen driver source code of this package?

sun6560032, 2022-03-29:
@wgarnett
> @salvacam ，完整图像=>https://www.mediafire.com/file/qg27qq7pllykwok/q8_od_fix.7z/file

Do you have the screen driver source code of this package?

salvacam, 2022-04-01:
> Do you have the screen driver source code of this package?

No, I don't have any code.
I just mixed the image of wgarnett and Stewrad

tiopex, 2022-04-21:
@wgarnett is key mapping working for you? On my m3, which is clone of q8, none of buttons working. How did you manage correct key mappings?

tiopex, 2022-04-22:
steps which I have done
1. compiled showkey from https://mirrors.edge.kernel.org/pub/linux/utils/kbd/ using arm-miyoo-linux-uclibcgnueabi-gcc from toolchain.7z
2. move compiled showkey to squashfs to /bin
3. add entry to /etc/inittab 
   ::sysinit:/bin/showkey > /mnt/keys.log 2>&1
4. create new squashfs and move to sd card
5. when console boot, I can get keycodes only for two keys L and start. Other buttons are not recognized

wgarnett, 2022-04-22:
@tiopex , I don't have M3 , you can try to modify  gmenu2x/input.conf  and test

tiopex, 2022-05-08:
@wgarnett do you have source code of video driver from below image?

> full image => https://www.mediafire.com/file/qg27qq7pllykwok/q8_od_fix.7z/file

TriForceX, 2022-05-09:
The file is in the steward-fu repository kernel source https://github.com/steward-fu/q8/releases

Apaczer, 2023-03-11:
duplicate of https://github.com/TriForceX/MiyooCFW/issues/242, moving to discuss