TriForceX/MiyooCFW #283: Unable to play any Pico8 Games

Timmmydx, 2022-01-10:
Hi Team, 

I just installed the CFW 1.3.3 on my powkiddy Q90. after that I tried downloading Pico8 games from their official website [https://www.lexaloffle.com/pico-8.php] but after transferring it to my Q90. all the games are unable to run. 

Any advice on what I should do :<

Timmmydx, 2022-01-12:
anyone having this same issue ?

salvacam, 2022-01-15:
If pico 8 doesn't work for you, you can try Retroarch's retro8 core.

Rezzy-dev, 2022-01-16:
You can also try Fake08 -- I had better luck with it:

[fake08.zip](https://github.com/TriForceX/MiyooCFW/files/7876354/fake08.zip)

(Use the Select or Start button -- I can't remember exactly -- to cycle through view modes.)

Note that Pico-8 support on Miyoo is still fairly new, and that the emulators are not up-to-date with the latest in Pico-8's development. They're behind a bit. So chances are that older games will work better than newer ones.

Timmmydx, 2022-01-17:
Hi @Rezzy-dev how do i add that file to my powkiddy ?
just unzip and drag in to the SD card ?

Rezzy-dev, 2022-01-17:
Unzip it, and if you look on your SD card, inside the "emus" folder, there is the "pico8" folder. Inside that you'll find a file named same as this one. Rename or delete that file (your choice) and copy this one to its place. Simple as that. :)

Rezzy-dev, 2022-01-17:
The big benefit you have with RetroArch is that the emulators are more up-to-date there, which is relevant when it comes to something still constantly in development, like the Pico-8 engine. But you do need to be a little more technically apt to use RetroArch, as it requires a bit of setting up.

Rezzy-dev, 2022-01-17:
Note: before you quit this emulator (Fake08), make sure you always "exit to menu" in the Pico-8 games, otherwise with some games you may experience the emulator freezing on you. If that ever happens, hold the Reset button and press the Select button. That will force quit the emulator and shut down the device cleanly. After the screen turns black, you can flip the power switch safely.

Timmmydx, 2022-01-20:
noted thanks @Rezzy-dev  will give it a try when im free

closed 2022-02-21 by Timmmydx