TriForceX/MiyooCFW #288: dosbox screen resize

wgarnett, 2022-01-17:
Hello , I have a question 

My Device is 320x240  , but when I play , game is work on 640x480 , so on Screen I only look 1/4 game screen
how to fix dosbox screen issue ???

salvacam, 2022-01-17:
https://github.com/TriForceX/MiyooCFW/issues/281#issuecomment-1014781542

wgarnett, 2022-01-22:
@salvacam 

this game , can you work fine ?  
http://dos-bin.zczc.cz/%E9%87%91%E5%BA%B8%E7%BE%A4%E4%BE%A0%E4%BC%A0.zip

Rezzy-dev, 2022-01-23:
What on earth is this? It's a mess. :eyes: 

You're not going to be able to run a late 90s DOS game on the Miyoo DOSBox, and especially not one of this size. DOSBox on Miyoo currently only supports lighter games, mostly from the late 80s, and up to 1992. Everything else, graphics and audio heavy games from the mid to late 90s, will run incredibly slow if they manage to load at all.

salvacam, 2022-01-23:
> @salvacam
> 
> this game , can you work fine ? http://dos-bin.zczc.cz/%E9%87%91%E5%BA%B8%E7%BE%A4%E4%BE%A0%E4%BC%A0.zip

What file should I run?
I am running the Z.COM file

Some splash screen is shown and then it is heard but nothing is shown.

https://user-images.githubusercontent.com/6243712/150684551-8cc91dc1-2705-4577-b0c6-bc58a8485da7.mp4

As you have been told dosbox only works well for very old games, from the 80s or early 90s at the most.

Rezzy-dev, 2022-01-23:
> Dosbox in Miyoo is only good for very old games, from the 80s or early 90s at the most.

Yep, more specifically games that run CGA and EGA video modes, mostly (late 80s games), and early 90s games that use VGA in 320 x 200 screen resolution. Some later DOS games don't support lower VGA screen resolutions. This may be the case here.

> What file should I run?
> I am running the Z.COM file

That's the one. Or "play.bat" -- it will give you the same result.

> Some splash screen is shown and then it is heard but nothing is shown.

After the logo, you're meant to see the game's title screen show up, but this isn't happening at the moment, because the game is trying to run at a higher screen resolution than what the Miyoo's screen supports.

See if you can change the game's screen resolution in the settings EXE/COM. But I suspect the resolution switching is automatic, and the game just doesn't support a 320 x 200 screen resolution.

Which means it won't run on the Miyoo.

Rezzy-dev, 2022-01-23:
Here are VGA's specs:

https://en.wikipedia.org/wiki/Video_Graphics_Array#Standard_graphics_modes

Rezzy-dev, 2022-01-23:
Miyoo devices can barely emulate 16-bit ROMs, let alone 32-bit DOS games with advanced graphics and sound. Even if the game did support a lower screen resolution, chances are it would run way too slow on the Miyoo device to be of any use.

wgarnett, 2022-01-24:
ok , I See , thanks everyone

Rezzy-dev, 2022-01-24:
Games I can recommend for DOSBox on the Miyoo are things like Alley Cat, Battle Chess, Ski or Die, Soko-Ban, Airborne Ranger, The Games: Summer Edition, North & South, and Budokan.

Very early 90s games such as Prehistorik, Prince of Persia, Pushover, Centurion: Defender of Rome, and Tank Wars are also playable, although with some slowdowns/performance issues.

wgarnett, 2022-01-24:
@Rezzy-dev  , thanks , I'll try the games you recommend

Rezzy-dev, 2022-01-24:
If you're going to use DOSBox, check out the setup and instructions I've provided here: https://github.com/TriForceX/MiyooCFW/discussions/268

This native setup allows for more customisation/setup for the games, using BAT files (text files that are Batch scripts in DOS).

I've included the setup for Alley Cat as an example that you can copy.

Budokan, for example, needs the following instructions/setup to run on Miyoo:

CD BUDOKAN
BUDO.COM ADLIB

Some games will attempt to use the PC speaker for music. There is a bug in DOSBox for the Miyoo that causes it to crash if too many complex sounds are played at once through the PC speaker (such as when it is used for music). So you want to use AdLib instead of the PC speaker for music where possible, and use the PC speaker for sounds only.

wgarnett, 2022-01-24:
@Rezzy-dev  ,  great , thanks your instructions