TriForceX/MiyooCFW #289: Powkiddy Q90 - stock firmware

Tekkno303, 2022-01-18:
I bought powkiddy Q90.

After starting, the console did not work. She paused at the start screen NxHope. It turned out that the software on the microsd card is damaged / incomplete. I have updatwd the new costom software cfw on other sd card. The console works very well on CFW.
However, I would like to have the original manufacturer's software (stock firmware (NxHope) on my card.
Will someone share the iso image of the card with the original manufacturer firmware NxHope?

Rezzy-dev, 2022-01-19:
Most likely the original card that your device came with is faulty. Mine was. They use hacked SD cards to make them appear larger than they are. NEVER use the default card that arrived bundled with your PowKiddy device -- it's not wise to -- but instead install a custom firmware on a new SD card. If you do use the bundled card, you risk losing your data.

As for NxHope, why would you want to revert to such an early version of the Miyoo CFW? I do still have a disk image of it, but it's 32 GB for a 32GB or larger card. Quite a hefty upload/download... and for what purpose? So you'll have a less functional, older release of the CFW?

zahinwadud, 2022-01-22:
My Q90 is on the way.
A question, does the custom firmware (e.g. Miyoo 1.3.3) come with installed ROMs? If not, then where can I find/download those ROMs?

Rezzy-dev, 2022-01-23:
> A question, does the custom firmware (e.g. Miyoo 1.3.3) come with installed ROMs?

No, you should save any ROMs you wish to keep to your computer. The custom firmware only comes with the software, not the ROMs.

Legally, ROM sharing is not allowed. You are supposed to be dumping your own ROMs from actual cartridges of the games you own for private use, or downloading ROMs of only the games you own. For this legal reason, you won't get an answer here as to where you can find and download ROMs online, but as most things online, they're shared and available.

Rezzy-dev, 2022-01-26:
https://github.com/NxHope