TriForceX/MiyooCFW #292: Asciiportal missing

RobinTPotter, 2022-01-22:
Hi - i noticed the original project is no longer maintained but that previously asciiportal has been available. I don't seem to be able to get it to show the menu

Rezzy-dev, 2022-01-23:
Try the button combination Start + Select. In some earlier software written for the Miyoo this is how you access the menu.

RobinTPotter, 2022-01-23:
Unfortunately not. Start works to change the portal display mode but sepect is mapped to a portal fire. Together they do nothing. The only way to stop the game (and switches off the pocket go) is to hold reset first then select. Very weird

Rezzy-dev, 2022-01-23:
Holding Reset and pressing Select is a system-swide command that executes a forced shutdown. It's intended for when programs freeze/hang and you can't quit them.

Rezzy-dev, 2022-01-23:
The old Asciiportal source code can be found here:

https://github.com/steward-fu/miyoo/releases/

Rezzy-dev, 2022-01-23:
The Select button (by itself) seems to be the quit button in Asciiportal, according to this code. (ESC/F1 are quit -- which are mapped to the Select button.)

Rezzy-dev, 2022-01-23:
Try the included "asciiportal" binary in the source package on your system.

RobinTPotter, 2022-01-24:
Thanks @Rezzy-dev I will take a look

RobinTPotter, 2022-01-25:
Hi @Rezzy-dev thanks for sharing the source link, I downloaded and extracted asciiportal.7z and the binary which was there works as expected. Thanks very much for your help, I might be able to get off level 19 now.

closed 2022-01-25 by RobinTPotter