TriForceX/MiyooCFW #296: Add a shutdown feature and password protect the hidden menus.

freaxeh, 2022-01-28:
for goodness sake. I keep having to press select and start together at the menu and this time I accidentally deleted the executable information for my favourite amiga emulator while trying to press select and start at the same time to get the shutdown screen to load.

It took me several hours last night too to figure out where I needed to copy the kernel.rom file and now I have to spend plenty of hours more having to repair the damage that I've done.

What would happen if a kid got a hold of this and did the exact same thing?

TriForceX, 2022-01-28:
Usually the "delete link" from select menu is just for the link... but yea, that link has settings inside, so it can be lost easily.

For a fast workaround for this issue i would like to add an extra confirmation to continue the deleting (and moving the item to the end of the menu)

Rezzy-dev, 2022-01-29:
I don't understand. Isn't the first item in the menu "Edit Link", not delete link? For me it enters the link editing menu, which you can safely exit by pressing the B button.

This is slight nitpicking, but can we also call them launchers, not links? The word "link" makes no sense in this context. They're not aliases, they're more than shortcuts, they're launchers for the programs, containing launch settings.

@freaxeh, you don't need to hold down both Select and Start anymore as the shortcut to shutting down. That was in the early releases of the Miyoo CFW (if ever). The actual shortcut to shutting down is simply holding down the Start button by itself.

Rezzy-dev, 2022-01-29:
Also, @freaxeh, if you wish to back up your launchers (links) for safety, copy what's in "gmenu2x/sections" on the SD card to your backup location.

Rezzy-dev, 2022-01-29:
A delete confirmation dialog would be very useful in GMenuNX for safe measure. Currently it is possible to delete an entire section of launchers/links (a whole category/page) accidentally.

Apaczer, 2022-01-30:
The safest way would be Reset+Select hotkey, which will shutdown console instantaneously without _shutting dialog_

Rezzy-dev, 2022-01-30:
> The safest way would be Reset+Select hotkey, which will shutdown console instantaneously without shutting dialog. I use it constantly on these handhelds, very convenient.

Reset + Select will execute force quit on all currently running applications and quickly shut down the system. It's meant to be used for emergencies, when emulators/software freeze. It's not a proper shutdown, but definitely better than just flicking the switch, It's a compromise between the two.

If you can shut down the system by quitting the currently running software/emulator and then holding Start or selecting "Power Off" in the Settings, do that.

Use Reset + Select only in emergencies, as it was designed for this use, not as a substitute for the normal shutdown. The danger is that some software may not deal well with a sudden force quit, and therefore certain software settings/files can get corrupted if you use a forced shutdown every time.

Reset + Select kills all currently running processes, without giving them the chance to cleanly exit -- and execute any exit tasks they may need to do to properly close the software.

Apaczer, 2022-02-08:
Thanks for clearing that out @Rezzy-dev, have been using that key combination blindly and I did not notice occasional fat32-checks (strangly currently running app/game had no issue after that force-closing and properly ended all its processes).

@freaxeh you might be interested in putting this _autoscript.sh_ file in your /main partition:
```
#!/bin/sh
cd /mnt/gmenu2x;
./gmenu2x > /dev/null 2>&1;

echo "Press Select, Pad o R to shutdown"
read -t 1 -n 1 check

if [ ! -z "$check" ]; then
    clear
    sync
    umount /dev/mmcblk0p4 > /dev/null 2>&1;
    poweroff
fi

exit

```

It will safely shutdown your console, whenever you close any app and press Select, D-Pad o Reset quickly afterwards. That is something @salvacam (thx!) came up with, looking for solution to this issue.

Rezzy-dev, 2022-02-09:
> Thanks for clearing that out @Rezzy-dev, have been using that key combination blindly and I did not notice occasional fat32-checks (strangly currently running app/game had no issue after that force-closing and properly ended all its processes).

In most instances you won't run into a problem, because software are generally designed so that their running process can be terminated at any time, and a sudden quit or crash won't ruin anything. However, if you happen to execute this key combination while the software is in the process of saving, for example (with or without your knowledge, in the background), the danger is that you can end up with a corrupted save (including software configuration files), and such that the FAT-32 check may not actually pick this corruption up. So the software may start to present problems from that point on without you even being aware that there is any problem with its files.

Additionally, some software may be designed in such a way that when they quit, they perform certain cleaning functions which will not end up running if you force quit them (if you terminate all running processes). Depending on how the software was programmed, this can, again, cause unforeseen problems the next time you run that software. This latter occurrance is rare, but nonetheless it's possible.

This is why software have a programmed user quit option instead of relying on the system to terminate all currently running processes (which is what Reset + Select does). So such a force quit should be reserved for when software hang/freeze, and you're not able to quit them or shut down the system normally/cleanly. At least terminating all running processes first ensures that the SD card is not being written to when you flip the power switch -- therefore minimising the risk of FAT32/data corruption.

salvacam, 2022-02-09:
> 

> Thanks for clearing that out @Rezzy-dev, have been using that key combination blindly and I did not notice occasional fat32-checks (strangly currently running app/game had no issue after that force-closing and properly ended all its processes).
> 
> @freaxeh you might be interested in putting this _autoscript.sh_ file in your /main partition:
> 
> ```
> #!/bin/sh
> cd /mnt/gmenu2x;
> ./gmenu2x > /dev/null 2>&1;
> 
> echo "Press Select, Pad o R to shutdown"
> read -t 1 -n 1 check
> 
> if [ ! -z "$check" ]; then
>     clear
>     sync
>     umount /dev/mmcblk0p4 > /dev/null 2>&1;
>     poweroff
> fi
> 
> exit
> ```
> 
> It will safely shutdown your console, whenever you close any app and press Select, D-Pad or Reset quickly afterwards. That is something @salvacam (thx!) came up with, looking for solution to this issue.

It will show the message "Press Select, Pad or R to shutdown" for a second if you don't press one of those buttons it returns to gmenu2x
The time can be modified on the line
read -t 1 -n 1 check
the value after -t is the seconds the message is displayed

If you want its operation to be the opposite, that when closing an emulator the console turns off if one of the buttons is not pressed in a second. You have to change the line
if [ ! -z "$check" ]; then
for
if [ -z "$check" ]; then