TriForceX/MiyooCFW #301: Misconfigured Max clock speed affects all Allwinner F1C100S devices

MrRocketFX, 2022-01-29:
The default max clock speed is set to 900mhz which will lock up the device if set to or above 798mhz.
The proposal is to set the "cpuMax" value to 798mhz to prevent lockups.
[image]

Rezzy-dev, 2022-01-30:
Here are the specs technical for the CPU:

https://linux-sunxi.org/F1C100s

Normal operating frequency is at 533 MHz, actually.

The maximum operating frequency it is able to perform at actually seems to vary from device to device:

https://www.rghandhelds.com/handheld-specs

(Some are marked at max 798 MHz, and others at max 702 MHz here.)

According to the original marketing/documentation, this processor supports a speed of up to 900 MHz, but in practice the maximum speed tends to vary from device to device, and to be lower, somewhere around 700 MHz.

I'm not sure if the maximum setting should be changed, as overclocking is always known to cause problems. You're taking a risk trying to overclock your processor in the first place. I think it may be better to allow for the full range rather than limit the user in the UI. We don't know if a particular device may actually support up to 900 MHz (which is the upper technical, physical performance limit of this CPU).

But if you wish to change it on your system, simply edit "gmenu2x/gmenu2x.conf" on your SD card and set the "cpuMax" setting in it to your desired value.

Apaczer, 2022-02-08:
https://github.com/gameblabla/bittboy_oc

You can check from above OC table, that anything beyond 798MHz have no practical usage. However users from first PocketGO line claims that overclocking above 732MHz may cause any running program to close on itself. Moreover one person from support stuff with Powkiddy V90, have stated that 828MHz should give best performance results (see PCSX-Rearmed Config Library). 
Can't say from where these mismatches come from for one CPU, but personally I've found 798MHz to be stable and effective OC value (on V90).

Rezzy-dev, 2022-02-09:
The mismatch comes from the fact that the hardware used in these devices, even along the same model/line in individual devices, varies. So the components that make up the motherboard, the memory, etc., and the work that was done to fit them together, may actually differ between to PocketGos or between two PowKiddy V90s.

In other words, no two devices are exactly the same, even if they carry the same processor. The maximum working speed of the processor is impacted by a lot of other factors. And just because the processor can run at 900 MHz does not mean that the whole system can.

Rezzy-dev, 2022-02-10:
These units are assembled in various factories in China, and multiple factories often work to produce the same model/product/brand while each doing their own thing. While basic quality assurance exists, there's no specification that states exactly what needs to go into each device (only rough specs) -- so if there is other stock that can be used, and that serves to save on manufacturing costs, you can sometimes end up with very different hardware in the same device, produced in two different factories.

Both devices will work well and will do what they're supposed to do, but when it comes to things like top performance, their behaviour may differ quite a bit due to the different internal components used in their manufacture -- despite the fact that they are both a PowKiddy V90, for instance.

Hope this helps to better understand it.

Apaczer, 2023-03-14:
cpuMax acceptable value differs from used components on PCB as @Rezzy-dev pointed out. It has been confirmed by a few users already, so there is no one and only MAX CPU for this SOC. 

I added CPU settings dialog on latest gmenu2x, so user can define those values by himself from personal testings also clock table has been severly changed see: 
https://github.com/MiyooCFW/gmenu2x/blob/1a6f11cf3777b7b8ec4ae37b87b9fbebc1a1c48b/src/platform/miyoo.h#L35-L62

closed 2023-03-14 by Apaczer

Baodejia, 2024-03-01:
Hello
I own 3 powkiddy Q90, with cfw 1.3.3 All can reach 900Mhz (max setting) but sometimes harder games close itself best frequency is 894Mhz.
With cfw 2.0 beta (uclib) and emulator (not retroarch) 1 console can reach 936Mhz but with harder games best is 912Mhz and the 2 other console work at 960Mhz (max setting 1008Mhz).
I can see the improvement with Virtua racing 32x and other system like GBA,Snes and FBA neogeo (beyond kof99) but with ps1 it seems not as good as cfw 1.3.3 with 894Mhz, i think the issue was the file system used in cfw 2 kill performance of PS1.
I own 1 V90 and max frequency is about 860-870Mhz
Sorry for bad english
PS: FBA can run larger size games, Gneo can't

Baodejia, 2024-03-01:
Hello
I just modified on 1 of my Q90 with cfw 1.3.3 the gmenu.conf file with 999Mhz max value, i can reach it with all emu but need to play some games to check stability and performance
Thanks Mr RocketFx