TriForceX/MiyooCFW #306: history played games

ahahahaah, 2022-02-12:
hello,
is there a way to see which games i have played lastly? a kind of history that we could check to find the game easily and replay it quickly rather than execute the emulator and browse the list to re-find them?
thanks

Apaczer, 2022-02-13:
Generally by default the gmenu2x frontend will point you to the last selected section, then emulator, then game (in folder's selector) you have played earlier - so it is just a matter of a couple "clicks". If you wish to see full history or recently played games, then there is RetroArch frontend where you will find _History_ list. 
You can even create separate Link in gmenu2x for launching recently played game - you can read more about it here: <https://github.com/TriForceX/MiyooCFW/wiki/RetroArch-Setup>
or here, where you can learn how to implement RetroArch as main frontend:
<https://jahed.dev/2022/01/25/improving-powkiddy-v90-ux/>

Rezzy-dev, 2022-02-15:
The current version of GMenuNX (the main UI interface on Miyoo) doesn't have such a listed history of what you've run in the past -- it doesn't track your activity. It only remembers the last selected item in a ROM/file list.

Such a history of the last/most selected games was a discussed potential feature for the future, but to my best knowledge no work has really been done on it, except for some experimental testing in the Beta Testing archive.