TriForceX/MiyooCFW #307: the pre-installed game "reminiscence" doesnt launch

ahahahaah, 2022-02-13:
hello
in the pre-installed games section there is reminiscence, but it doesnt launch,
i have checked the folder and there is a "reminiscence.elf" file (842 kb) and an empty "data" folder
 is it intentionaly not installed completely? or is it an other problem?

Rezzy-dev, 2022-02-15:
Some games will require additional assets for you to install -- as they are open source implementations of copyrighted, purchasable content. So you need to buy the game and copy the original game assets into it in order to be able to run the open-source implementation/version of it. This may be the case here, too.

ahahahaah, 2022-02-15:
ok i didn't know that
thanks

ahahahaah, 2022-02-19:
is there a "clean" way to uninstall some pre-installed games?
delete the folders is ok or is there a special process?

Rezzy-dev, 2022-02-20:
Deleting (or moving) the folder off the SD card is all you need to do to uninstall the preinstalled software.

If you want to also clean up any trace of it left behind, you'll need to enable seeing hidden files on the SD card, and check the root on the "main" partition. There should be a folder there named the same as the software you've deleted. This contains the software's saves and settings. Delete (or move) that folder.

closed 2023-03-14 by Apaczer