TriForceX/MiyooCFW #309: picodrive emulator

ahahahaah, 2022-02-16:
Hello,
With the megadrive emulator "picodrive" i can't set the display to fullscreen, in this page https://github.com/TriForceX/MiyooCFW i haven't found it in the emulators list, so i can't contact the developer to ask him if there is a way to have it. Maybe in editing a line of a file?
When i clic the menu/reset button during a game, the background image becomes in fullscreen  (what i want) but in the display settings there is only one display choice, and when i leave the emulator menu page the image goes back in standard mode.
thx

Apaczer, 2022-02-16:
<https://github.com/irixxxx/picodrive/releases>

You will find above the latest revision of that emu for Miyoo (delete old config file to make it work). If we're talking about Mega Drive emulation it has _vertical scaling_ option in the _display_ menu which setting it to _software_ should stretch your image to full screen.

ahahahaah, 2022-02-16:
thanks i have done it. 
indeed i had to delete the config file but if i press a key for a function (for example for turbo A) how can i delete/reset the function from the key i have chosen, i mean to leave this function with a blank/void choice?
i ask this because if i choose a key, it means i can't use it for an other function otherwise there will be 2 different functions for 1 key pressed.

Apaczer, 2022-02-16:
you can't from knowledge - the _clear_ bind key is assigned to ";" corresponding keyboard button which is in no way binded in devices kernel to any physical button. Your best option would be then to delete again config file, or edit it manually through text editor.

ahahahaah, 2022-02-17:
ok i will try this 
thanks

jSTE0, 2022-02-18:
I submitted irixxxx/picodrive#61 to add Miyoo support, with the correct button mappings.
Until that is merged, I've built a binary from the current HEAD and made it available at https://github.com/jSTE0/picodrive/releases.

ahahahaah, 2022-02-18:
that's great!
thank you for your efforts and commitment :D

Apaczer, 2022-02-19:
@jSTE0 that was indeed lacking feature, however I saw that the _Emulator controls_ in this build can not be binded or unbinded through main menu (so a few buttons are left hardcoded).

jSTE0, 2022-02-19:
> @jSTE0 that was indeed lacking feature, however I saw that the _Emulator controls_ in this build can not be binded or unbinded through main menu (so a few buttons are left hardcoded).

Good spot, I did a quick port and didn't check that menu. It looks like the emulator controls page is just broken, key bindings are not populated and it just uses whatever was hardcoded, regardless of what you try to configure. You could just edit the config file directly and that should work as the default key for the menu is "Escape" which is actually the Select button whereas it has been configured to R in the original config file.